nerves-project/nerves #4: Support for Elixir 1.3.0-dev

bruce, 2016-05-21:
I tried building nerves-examples/blinky -- and bootstrapping a new project -- with Elixir 1.3.0-dev (HEAD).

When building blinky, I got the following error (trying both `rpi2` and `rpi3`):

```
DEBUG=1 mix firmware
warning: the archive nerves_bootstrap requires Elixir "~> 1.2.4 or ~> 1.3" but you are running on v1.3.0-dev
Nerves Precompile Start
Nerves Env current
Nerves Env loaded
Nerves Precompile End
===> Expanded command sequence to be run: []
===> Expanded command sequence to be run: [{default,app_discovery},
                                                  {bare,compile}]
===> Evaluating config script "/Users/bruce/Development/nerves-examples/blinky/deps/rpi3/erlware_commons/rebar.config.script"
===> Uncaught error in rebar_core. Run with DEBUG=1 to see stacktrace
===> Uncaught error: {badmatch,
                             {error,
                              {19,file,
                               {error,
                                {badmatch,
                                 ["3","0","0+build","272","ref7d29b74"]},
                                [{erl_eval,expr,3,[]}]}}}}
===> Stack trace to the error location: [{rebar_config,consult_file_,
                                                 1,
                                                 [{file,
                                                   "/home/travis/build/rebar/rebar3/_build/default/lib/rebar/src/rebar_config.erl"},
                                                  {line,106}]},
                                                {rebar_config,consult_file,1,
                                                 [{file,
                                                   "/home/travis/build/rebar/rebar3/_build/default/lib/rebar/src/rebar_config.erl"},
                                                  {line,90}]},
                                                {rebar_app_discover,
                                                 project_app_config,2,
                                                 [{file,
                                                   "/home/travis/build/rebar/rebar3/_build/default/lib/rebar/src/rebar_app_discover.erl"},
                                                  {line,127}]},
                                                {rebar_app_discover,
                                                 merge_deps,2,
                                                 [{file,
                                                   "/home/travis/build/rebar/rebar3/_build/default/lib/rebar/src/rebar_app_discover.erl"},
                                                  {line,81}]},
                                                {rebar_app_discover,
                                                 '-do/2-fun-1-',5,
                                                 [{file,
                                                   "/home/travis/build/rebar/rebar3/_build/default/lib/rebar/src/rebar_app_discover.erl"},
                                                  {line,46}]},
                                                {lists,foldl,3,
                                                 [{file,"lists.erl"},
                                                  {line,1262}]},
                                                {rebar_prv_app_discovery,do,
                                                 1,
                                                 [{file,
                                                   "/home/travis/build/rebar/rebar3/_build/default/lib/rebar/src/rebar_prv_app_discovery.erl"},
                                                  {line,38}]},
                                                {rebar_core,do,2,
                                                 [{file,
                                                   "/home/travis/build/rebar/rebar3/_build/default/lib/rebar/src/rebar_core.erl"},
                                                  {line,125}]}]
```

When attempting to bootstrap a new project with `mix nerves.new nerves_test --target rpi3`, I see:

```
warning: the archive nerves_bootstrap requires Elixir "~> 1.2.4 or ~> 1.3" but you are running on v1.3.0-dev
** (Mix) Nerves v0.1.2 requires at least Elixir v1.2.4.
 You have 1.3.0-dev. Please update accordingly
```

I'd say the `~> 1.3` requirement was just too strict, but the error in building blinky (fixed by using 1.2.5) seems to point out a deeper issue supporting 1.3.0-dev.

bruce, 2016-05-21:
Working on the blinky build with @asonge on Slack, I pinned `erlware_commons` to a more recent version:

```
{:erlware_commons, "~> 0.21.0", override: true}
```

I got through `mix compile` this time on `1.3.0-dev`, but got the following when running `mix firmware`:

```
$ mix firmware
warning: the archive nerves_bootstrap requires Elixir "~> 1.2.4 or ~> 1.3" but you are running on v1.3.0-dev
Nerves Firmware Assembler
** (RuntimeError)   Environment variable $NERVES_SYSTEM is not set

    lib/mix/tasks/firmware.ex:18: Mix.Tasks.Firmware.run/1
    (mix) lib/mix/cli.ex:58: Mix.CLI.run_task/2
    (elixir) lib/code.ex:363: Code.require_file/2
```

mobileoverlord, 2016-05-21:
Thanks for the investigation @asonage @bruce. I verified that its failing to execute mix firmware at the step you mentioned. I guess in the meantime we can force elware_commons

mobileoverlord, 2016-05-21:
fixed by
nerves-project/nerves@a01596ec59dee7a0697135fe9efa85abdad6b234

closed 2016-05-21 by mobileoverlord