nerves-project/nerves #7: Timestamps on files burned to SD card are off by 1 month

wsmoak, 2016-05-30:
I noticed that after `mix firmware.burn`, the timestamps on the files on the SD card are showing in Finder as **April** 30, 2016 at 12:59 AM when it was burned on **May** 29, 2016 at 8:59 PM Eastern.

The day and the time (in UTC) are correct, but the month is off by one.

The timestamp on the .fw file under _images is correct:
```
$ ls -l _images/rpi2/
-rw-r--r--  1 wsmoak  staff  19045757 May 29 20:59 delete_me7.fw
```

Elixir 1.2.5
Nerves 0.3.2
fwup 0.6.0  (apparently there is an 0.6.1 but I haven't been able to get it from homebrew)

fhunleth, 2016-05-30:
This has got to be a fwup issue. I use a 3rd party FAT filesystem library, and it really sounds like I made an off-by-one error when integrating it. I'll check it out. Thanks!

fhunleth, 2016-05-30:
I verified the issue and moved it to `fwup`. Thank you so much for catching this!.

closed 2016-05-30 by fhunleth