nerves-project/nerves #25: rootfs additions in deps?

arjan, 2016-07-19:
It would be handy when deps can specify rootfs additions as well. Currently only the toplevel project can do this, but libraries may want to add things to the root filesystem as well, e.g. libraries (in `/lib`), executables or config files.

How this would be configured is not really clear to me yet, maybe every application can expose a `:nerves_rootfs_additions` config key; the mix task would then need to scan all application configuration for this key:

    config :my_library, :nerves_rootfs_additions, "my-rootfs-extra/"

Which the mix firmware task would then translate into multiple `-a <path>` arguments to the `rel2fs.sh` script.

cc @fhunleth

mobileoverlord, 2016-07-19:
For this to be handled, we would have to require that the developer knows that an included dep has rootfs-additions and would need to explicitly add it to their config in their app. So for example

`:my_cool_lib` has rootfs additions as well as the main project we could all a pass of 
```
config :nerves, :firmware,
  rootfs_additions: [:my_cool_lib, :my_app]
```
In this case, all apps would have to have a known location for where nerves rootfs additions would exist and or a way of overriding. We could set this in the env in the mix file.

We would also have to have an elegant way of handling merge conflicts. If you have two deps which are trying to write to the same rootfs file. I am open for a discussion on how to handle this.

arjan, 2016-07-19:
Something like this makes sense too, indeed. Maybe just a topfolder called `rootfs-additions`?

About the merge conflicts: right now it does `cp -R`, if we change that to `cp -Ri`, it'll prompt you when it overwrites a file while building the final rootfs tree. Only this way you wouldn't know which library put that file there first, which could be hard. Buildroot has something for that, iirc, it writes a text file somewhere which administers which files gets put in the root fs.

josevalim, 2016-08-31:
Since dependencies are likely in deps/, I would just allow multiple directories to be given as rootfs additions. This way you don't need to directly support dependencies but rather get that for free once you have multiple directories.

Regarding conflicts, you can either warn when building it or explicitly raise and ask the user to configure a map choosing how conflicts should be resolved:

    rootfs_overrides: %{"/etc/iex.exs" => "deps/foo/bar"}

fhunleth, 2020-10-03:
This is a really old bug. Rather than trying to shoehorn libraries so that they can install files in global directories like `/etc`, `/usr`, etc., we've taken the approach of setting paths to `priv` directories. This is more work, but gets rid of the issue of having libraries dump files to arbitrary places on the filesystem and the difficulties associated with figuring out who owns what, detecting conflicts, etc.

closed 2020-10-03 by fhunleth