nerves-project/nerves #29: Support NERVES_COMPILER_CACHE=local

bendiken, 2016-08-07:
Is there perhaps an undocumented equivalent of `NERVES_SYSTEM_CACHE=local` for the compiler toolchain (nerves_toolchain) download?

Right now, if I do the following:

    export NERVES_SYSTEM_CACHE=local
    NERVES_TARGET=qemu_arm mix do deps.get, deps.precompile
    NERVES_TARGET=rpi mix do deps.get, deps.precompile
    NERVES_TARGET=rpi3 mix do deps.get, deps.precompile

...the actual systems get cached fine, but the toolchain always gets re-downloaded for each target in each Nerves project:

    ==> nerves_toolchain
    Compiling 2 files (.ex)
    Generated nerves_toolchain app
    ==> nerves_toolchain_arm_unknown_linux_gnueabihf
    Compiling 1 file (.ex)
    Generated nerves_toolchain_arm_unknown_linux_gnueabihf app
    [nerves_toolchain][compile]
    [nerves_toolchain][http] Downloading Toolchain
    [nerves_toolchain][http] Toolchain Downloaded
    [nerves_toolchain][http] Unpacking Toolchain
    ==> nerves_system_rpi3
    Generated nerves_system_rpi3 app
    [nerves_system][compile]
    [nerves_system][local] Checking Cache for nerves_system_rpi3-0.6.0
    nerves_system_rpi3-0.6.0 was not found in your cache.
    cache dir: /Users/arto/.nerves/cache/system
    Would you like to download the system to your cache? [Yn] 
    [nerves_system][http] Downloading System
    [nerves_system][http] System Downloaded
    [nerves_system][http] Unpacking System

I'd like to be able to work on new Nerves projects offline and/or on data-limited mobile connectivity. It seems like caching everything, one way or another, would be a good idea to enable such a work modality.

mobileoverlord, 2016-08-07:
I have this functionality staged in a separate branch and I am working on releasing this near ElixirConf. I'll message you back when this is available.

bendiken, 2016-08-07:
@mobileoverlord Awesome, looking forward!

mobileoverlord, 2016-12-01:
Merged in master

closed 2016-12-01 by mobileoverlord

bendiken, 2016-12-01:
Thanks a bunch, @mobileoverlord! Looking forward to testing it.