nerves-project/nerves #33: Document ssh-askpass dependency on Linux

bendiken, 2016-08-09:
On @DanKomorny's fresh Lubuntu 16.04 installation:

    $ NERVES_TARGET=rpi mix firmware.burn
    Use 7.4 GiB memory card found at /dev/sdb? [Yn] y
    sudo: unable to run /usr/bin/ssh-askpass: No such file or directory

The relevant bit of code is at: https://github.com/nerves-project/nerves/blob/d8e6bc217bde030813c88d4ea05b0782903c069b/lib/mix/tasks/firmware.burn.ex#L47

Resolved by `sudo apt-get install ssh-askpass`.

The Nerves [installation guide](https://hexdocs.pm/nerves/installation.html) should probably document this requirement on Linux?

mobileoverlord, 2016-08-10:
Sounds good. I think that this host requirement should be in the docs. Do you want me to add it or can you make a separate PR for this? :)

Thanks!

bendiken, 2016-08-11:
Sure thing, I'll send a pull request in a bit.

GregMefford, 2016-08-13:
@bendiken: I went ahead and fixed it. Thanks for pointing out the missing step! :balloon:

closed 2016-08-13 by GregMefford

bendiken, 2016-08-13:
Thanks, @GregMefford! However, note that the [getting started guide](https://hexdocs.pm/nerves/getting-started.html) never mentions `sudo`, so the text you added is perhaps somewhat misleading as-is given that we aren't actually ever asking people to use `sudo mix firmware.burn`.

GregMefford, 2016-08-13:
@bendiken: good point. Maybe it's simpler and more consistent to just say it's required. I've personally never noticed that it's missing because I just do `sudo mix firmware.burn` :grinning:

GregMefford, 2016-08-13:
Updated!

bendiken, 2016-08-13:
Looks good 👍

craigbrett17, 2018-10-17:
Helped me with a completely unrelated TeamCity SSH issue that was a pain to Google, I didn't know it was an apt package. Thanks!

drannex42, 2022-03-27:
I know this is several years old, but for anyone else who may end up googling for a solution to `sudo: unable to run /usr/bin/ssh-askpass: No such file or directory`

You need to create a symbolic link between ssh-askpass and your existing askpass installation.
```
ln -s $SSH_ASKPASS /usr/bin/ssh-askpass

# or 

ln -s /location/to/askpass /usr/bin/ssh-askpass

```

If the above didn't work, check to see if you have $SSH_ASKPASS set by running `echo $SSH_ASKPASS` in your terminal. 

If you don't, then you need to either install ssh-askpass (ubuntu is already listed in the docs), or if you are using Fedora, install the gnome askpass with `sudo dnf install gnome-ssh-askpass` and rerun the above.

commit 44ae316f (2016-08-13): Recommend installing `ssh-askpass` on Linux

Resolves #33.