nerves-project/nerves #39: mix firmware fails with poor error message when NERVES_TOOLCHAIN points to a directory that no longer exists

josevalim, 2016-08-31:
I did the mistake of removing the directory that NERVES_TOOLCHAIN points to and got a bad error message.

My config:

```
NERVES_TOOLCHAIN=/Users/jose/OSS/badge/pen/nerves-mipsel-unknown-linux-musl-darwin-x86_64-v0.6.3
NERVES_SYSTEM=/Users/jose/OSS/badge/pen/nerves_system_linkit
```

The error:

```
$ mix firmware
** (FunctionClauseError) no function clause matching in String.replace_suffix/3
    (nerves_bootstrap) lib/mix/tasks/nerves.loadpaths.ex:20: Mix.Tasks.Nerves.Loadpaths.run/1
    (nerves_bootstrap) lib/mix/tasks/nerves.precompile.ex:19: Mix.Tasks.Nerves.Precompile.run/1
    (mix) lib/mix/task.ex:296: Mix.Task.run_task/3
    (mix) lib/mix/task.ex:328: Mix.Task.run_alias/3
    (mix) lib/mix/task.ex:261: Mix.Task.run/2
    (mix) lib/mix/tasks/deps.compile.ex:62: Mix.Tasks.Deps.Compile.compile/2
    (mix) lib/mix/tasks/deps.loadpaths.ex:83: Mix.Tasks.Deps.Loadpaths.deps_check/2
    (mix) lib/mix/tasks/deps.loadpaths.ex:27: Mix.Tasks.Deps.Loadpaths.run/1
```

It may be worth checking if the directory in the environment variable actually exists before proceeding or something of sorts. :D

mobileoverlord, 2016-09-07:
I like it! thanks :)

mobileoverlord, 2016-09-12:
fixed in pkg
https://github.com/nerves-project/nerves/commit/3a1a1b932e5adc2bbe5d6dfe47f61332cb7887c5

closed 2016-09-12 by mobileoverlord