nerves-project/nerves #44: mix firmware doesn't properly rebuild release

ghitchens, 2016-09-07:
Repro:

- clone a simple nerves project 
- mix firmware
- change the mixfile to have a different version for the app
- mix firmware again

BUG:  an "upgrade release" is created but the OLD release is included in the firmware bundle.
```
gmb:stem garth$ mix firmware
Building release with MIX_ENV=dev.

You have dependencies (direct/transitive) which are not in :applications!
The following apps should be added to :applications in mix.exs:

        nerves_system_rpi3 -> nerves_system -> porcelain                                       => porcelain is missing from nerves_system
        nerves_system_rpi3 -> nerves_system_br                                                 => nerves_system_br is missing from nerves_system_rpi3
        nerves_system_rpi3 -> nerves_toolchain_arm_unknown_linux_gnueabihf -> nerves_toolchain => nerves_toolchain is missing from nerves_toolchain_arm_unknown_linux_gnueabihf
        logger_multicast_backend                                                               => logger_multicast_backend is missing from stem

==> Modifying release for Nerves System rpi3
==> The release for stem-0.2.1-pre-6a is ready!
==> You can boot a console running your release with `$ rel/stem/bin/stem console`
Updating base firmware image with Erlang release...
Parallel mksquashfs: Using 4 processors
Creating 4.0 filesystem on /Users/garth/nerves/stem/_nerves-tmp/combined.squashfs, block size 131072.
```

mobileoverlord, 2016-09-07:
I'll take a look at this today. I'll ping you afterward

mobileoverlord, 2017-01-05:
I've tested this in 0.4.3 and it seems that the issue has been resolved by forcing the release to clean before producing a new one. Additionally, we are going to be making this better upstream in erlinit through this issue: https://github.com/nerves-project/erlinit/issues/10

closed 2017-01-05 by mobileoverlord