nerves-project/nerves #45: Improve error message with toolchain can't be downloaded

fhunleth, 2016-09-12:
The problem here is that the toolchain doesn't exist in the GitHub release. Here's the current error message:

```
pi@raspberrypi:~/nerves-examples/hello_gpio $ mix compile
==> nerves_toolchain_arm_unknown_linux_gnueabihf
Compiling 1 file (.ex)
Generated nerves_toolchain_arm_unknown_linux_gnueabihf app
[nerves_toolchain][compile]
[nerves_toolchain][http] Downloading Toolchain
could not compile dependency :nerves_toolchain_arm_unknown_linux_gnueabihf, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves_toolchain_arm_unknown_linux_gnueabihf", update it with "mix deps.update nerves_toolchain_arm_unknown_linux_gnueabihf" or clean it with "mix deps.clean nerves_toolchain_arm_unknown_linux_gnueabihf"
** (RuntimeError) Nerves Toolchain Github cache returned error: "httpc request failed with: {:bad_status_code, 404}"
    (nerves_toolchain) lib/mix/tasks/compile.nerves_toolchain.ex:70: Mix.Tasks.Compile.NervesToolchain.cache/2
    (nerves_toolchain) lib/mix/tasks/compile.nerves_toolchain.ex:38: Mix.Tasks.Compile.NervesToolchain.run/1
    (mix) lib/mix/task.ex:296: Mix.Task.run_task/3
    (elixir) lib/enum.ex:1184: Enum."-map/2-lists^map/1-0-"/2
    (elixir) lib/enum.ex:1184: Enum."-map/2-lists^map/1-0-"/2
    (mix) lib/mix/tasks/compile.all.ex:19: anonymous fn/1 in Mix.Tasks.Compile.All.run/1
    (mix) lib/mix/tasks/compile.all.ex:37: Mix.Tasks.Compile.All.with_logger_app/1
    (mix) lib/mix/task.ex:296: Mix.Task.run_task/3

pi@raspberrypi:~/nerves-examples/hello_gpio $ 
```

It would be nice if it at least said what it was trying to download.

mobileoverlord, 2016-09-13:
I'll take a peek at this once pkg gets merged down. This is addressed with the consolidation of providers.

mobileoverlord, 2016-10-17:
this is fixed in master

closed 2016-10-17 by mobileoverlord