nerves-project/nerves #47: [documentation] 'targets' link is broken

thom-nic, 2016-09-20:
The [getting started guide](https://github.com/nerves-project/nerves/blob/master/docs/Getting%20Started.md) has several links to 
https://github.com/nerves-project/nerves/blob/master/docs/targets.html#target-dependencies

which appear to be broken - at least they are when viewing on Github.  Maybe when they are published they are correct?

mobileoverlord, 2016-09-20:
The docs are published on hexdocs which preprocesses the markdown and generates html pages out of them.

You can find them at https://hexdocs.pm/nerves/getting-started.html

Because of this, the links are set using relative page.html format. I do not have any ideas on how to support navigation through both github and hexdocs but i'll see if I can dig anything up other than direct linking.

mobileoverlord, 2016-09-20:
I don't see a clever way of supporting this especially because the extension difference between the two systems. When navigating on GH, you would need the extension to be MD and on Hexdocs you would need the html.

I "solved this" by adding a README to the docs folder with a link to the hex docs.
https://github.com/nerves-project/nerves/commit/99fb33c4f9e0f772509c9d57fcf897e0f796d154

closed 2016-09-20 by mobileoverlord

thom-nic, 2016-09-20:
Great, thanks!