nerves-project/nerves #57: Failing to download artifact from URL results in CHECKSUM Error

mobileoverlord, 2016-11-17:
### Environment
* Erlang version (erl -eval 'erlang:display(erlang:system_info(otp_release)), halt().'  -noshell):
* Elixir version (elixir -v):
* Operating system:
* Nerves Environment Info:
  (mix nerves.env --info) or
  General info about target / system / toolchain
### Current behavior
```[Nerves.Package.Providers.HTTP]   Downloading Artifact:
      nerves_toolchain_aarch64_unknown_linux_gnueabi-0.8.0-rc.0.darwin-x86_64.tar.xz
  Location
      https://github.com/nerves-project/toolchains/releases/download/v0.8.0-rc.0/nerves_toolchain_aarch64_unknown_linux_gnueabi-0.8.0-rc.0.darwin-x86_64.tar.xz

[Nerves.Package.Providers.HTTP] No Available Locations
** (File.Error) could not write to file "/Users/jschneck/Developer/nerves/toolchains/nerves_toolchain_aarch64_unknown_linux_gnueabi/.nerves/artifacts/nerves_toolchain_aarch64_unknown_linux_gnueabi-0.8.0-rc.0.darwin-x86_64/CHECKSUM": no such file or directory
    (elixir) lib/file.ex:716: File.write!/3
    lib/mix/tasks/compile.nerves_package.ex:23: Mix.Tasks.Compile.NervesPackage.run/1
    (mix) lib/mix/task.ex:294: Mix.Task.run_task/3
    (elixir) lib/enum.ex:1231: Enum."-map/2-lists^map/1-0-"/2
    (elixir) lib/enum.ex:1231: Enum."-map/2-lists^map/1-0-"/2
    (mix) lib/mix/tasks/compile.all.ex:19: anonymous fn/1 in Mix.Tasks.Compile.All.run/1
    (mix) lib/mix/tasks/compile.all.ex:37: Mix.Tasks.Compile.All.with_logger_app/1
    (mix) lib/mix/task.ex:294: Mix.Task.run_task/3```
Include code samples, errors and stacktraces if appropriate.

### Expected behavior
The error should be handled and stop before trying to initiate the CHECKSUM

fhunleth, 2016-11-17:
Are you really using aarch64???? Asking for a friend.

mobileoverlord, 2016-11-17:
no. I was just using it as a test when I noticed that I didn't like the error handling. It just that its the first toolchain in the repo when sorted alpha.

mobileoverlord, 2016-12-15:
Fixed in master

closed 2016-12-15 by mobileoverlord