nerves-project/nerves #60: mix firmware produces "no case clause matching: :eacces" error

cblock, 2016-12-12:
### Environment
* Elixir version (elixir -v):
Erlang/OTP 19 [erts-8.1] [source] [64-bit] [smp:4:4] [async-threads:10] [hipe] [kernel-poll:false] [dtrace]
Elixir 1.3.4

* Operating system:
mac OS Sierra 10.12.1

* Nerves Environment Info:
  (mix nerves.env --info) or
  General info about target / system / toolchain

### Current behavior
```
$  mix nerves.new tmp --target rpi
* creating tmp/config/config.exs
* creating tmp/lib/tmp.ex
* creating tmp/test/test_helper.exs
* creating tmp/test/tmp_test.exs
* creating tmp/rel/vm.args
* creating tmp/rel/.gitignore
* creating tmp/.gitignore
* creating tmp/mix.exs
* creating tmp/README.md

$ mix deps.get
Running dependency resolution
Dependency resolution completed
  bbmustache: 1.0.4
  cf: 0.2.2
  erlware_commons: 0.22.0
  exrm: 1.0.8
  getopt: 0.8.2
  nerves: 0.3.4
  nerves_system: 0.1.8
  nerves_system_br: 0.7.0
  nerves_system_rpi: 0.7.0
  nerves_toolchain: 0.7.1
  nerves_toolchain_armv6_rpi_linux_gnueabi: 0.7.2
  nerves_toolchain_ctng: 0.7.1
  providers: 1.6.0
  relx: 3.22.0
* Getting nerves (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves-0.3.4.tar)
  Using locally cached package
* Getting nerves_system_rpi (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves_system_rpi-0.7.0.tar)
  Using locally cached package
* Getting nerves_system (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves_system-0.1.8.tar)
  Using locally cached package
* Getting nerves_system_br (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves_system_br-0.7.0.tar)
  Using locally cached package
* Getting nerves_toolchain_armv6_rpi_linux_gnueabi (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves_toolchain_armv6_rpi_linux_gnueabi-0.7.2.tar)
  Using locally cached package
* Getting nerves_toolchain (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves_toolchain-0.7.1.tar)
  Using locally cached package
* Getting nerves_toolchain_ctng (Hex package)
  Checking package (https://repo.hex.pm/tarballs/nerves_toolchain_ctng-0.7.1.tar)
  Using locally cached package
* Getting exrm (Hex package)
  Checking package (https://repo.hex.pm/tarballs/exrm-1.0.8.tar)
  Using locally cached package
* Getting relx (Hex package)
  Checking package (https://repo.hex.pm/tarballs/relx-3.22.0.tar)
  Using locally cached package
* Getting bbmustache (Hex package)
  Checking package (https://repo.hex.pm/tarballs/bbmustache-1.0.4.tar)
  Using locally cached package
* Getting cf (Hex package)
  Checking package (https://repo.hex.pm/tarballs/cf-0.2.2.tar)
  Using locally cached package
* Getting erlware_commons (Hex package)
  Checking package (https://repo.hex.pm/tarballs/erlware_commons-0.22.0.tar)
  Using locally cached package
* Getting getopt (Hex package)
  Checking package (https://repo.hex.pm/tarballs/getopt-0.8.2.tar)
  Using locally cached package
* Getting providers (Hex package)
  Checking package (https://repo.hex.pm/tarballs/providers-1.6.0.tar)
  Using locally cached package

$ mix firmware
Building release with MIX_ENV=dev.

You have dependencies (direct/transitive) which are not in :applications!
The following apps should be added to :applications in mix.exs:

        nerves_system_rpi -> nerves_system                                                     => nerves_system is missing from nerves_system_rpi
        nerves_system_rpi -> nerves_system_br                                                  => nerves_system_br is missing from nerves_system_rpi
        nerves_system_rpi -> nerves_toolchain_armv6_rpi_linux_gnueabi -> nerves_toolchain      => nerves_toolchain is missing from nerves_toolchain_armv6_rpi_linux_gnueabi
        nerves_system_rpi -> nerves_toolchain_armv6_rpi_linux_gnueabi -> nerves_toolchain_ctng => nerves_toolchain_ctng is missing from nerves_toolchain_armv6_rpi_linux_gnueabi
        nerves                                                                                 => nerves is missing from tmp

==> Modifying release for Nerves System rpi
** (CaseClauseError) no case clause matching: :eacces
    lib/exrm/utils/utils.ex:110: ReleaseManager.Utils.relx/5
    lib/mix/tasks/release.ex:339: anonymous fn/4 in Mix.Tasks.Release.do_release/1
    lib/ex_unit/capture_io.ex:146: ExUnit.CaptureIO.do_capture_io/2
    lib/ex_unit/capture_io.ex:119: ExUnit.CaptureIO.do_capture_io/3
    lib/mix/tasks/release.ex:338: Mix.Tasks.Release.do_release/1
    lib/mix/tasks/release.ex:78: Mix.Tasks.Release.do_run/1
    (mix) lib/mix/task.ex:296: Mix.Task.run_task/3
    lib/mix/tasks/firmware.ex:27: Mix.Tasks.Firmware.run/1
```

cblock, 2016-12-12:
Manually downgrading the following dependencies in `mix.lock` fixes the problem for me:
``` 
* relx 3.22.0 -> 3.21.1
* erlware_commons 0.22.0 -> 0.21.0
* cf 0.22.0 -> 0.21.0
* nerves_system 0.1.8 -> 0.1.6
```

mobileoverlord, 2016-12-12:
I was going to fix this issue by merging in the distillery branch now that our PR's were mergesd and that 1.0.0 is out. I am going to be working on this today. New versions of everything should be out today / tomorrow. Bye bye relx...

cblock, 2016-12-12:
Ok, great. As soon as the new versions are out, I'll check if the bug is gone and close this bug accordingly.

mobileoverlord, 2016-12-15:
The lastest version of nerves has been release which provides support for distillery. Currently, the hex package has an issue with the template file missing to produce the release config. Can you please try using `{:nerves, github: "nerves-project/nerves"}`.

cblock, 2016-12-15:
Just tried it and it works like a charm. Thanks a lot!

closed 2016-12-15 by cblock