nerves-project/nerves #63: Question: how to autorun something

yizhiren, 2017-01-05:
hi ,

is there an official way to auto run something when system boot.
run a elixir program or a script file.
-------------------
i checked the online document, but i did't find a section say this.

thanks!

mobileoverlord, 2017-01-05:
Hi @yizhiren. With nerves, your application will start on boot of the system. To start additional processes or applications, you can call them at the time your application starts.

Here is an example of starting a process worker in elixir on application start
https://github.com/mobileoverlord/blinky_firmata/blob/master/lib/blinky_firmata.ex#L11

Here is an example of loading a kernel module when your application starts.
https://github.com/mobileoverlord/badge/blob/master/apps/badge_fw/lib/badge_fw.ex#L9

Hopefully this clears up a strategy for you. We also spend a lot of time on elixir-lang/nerves on slack

closed 2017-01-05 by mobileoverlord

yizhiren, 2017-01-06:
@mobileoverlord  Thanks!