nerves-project/nerves #65: Low timeout for Nerves.Package.Providers.HTTP

pinya, 2017-01-25:
### Environment
* Erlang version: 19
* Elixir version: 1.4.0
* Operating system: macOS 10.12.3

### Current behaviour

I'm working with not so fast internet connection (~5mbit/s) and cannot install `nerves_system_rpi`. It always fails after 120 seconds. 

```
mix deps.compile nerves_system_rpi
==> nerves_system_rpi
Generated nerves_system_rpi app
[Nerves.Package.Providers.HTTP]   Downloading Artifact:
  nerves_system_rpi-0.9.1.armv6_rpi_linux_musl.tar.gz
  From Location:
  https://github.com/nerves-project/nerves_system_rpi/releases/download/v0.9.1/nerves_system_rpi-v0.9.1.tar.gz

|=======================                           | 46% (47 / 102) MBcould not compile dependency :nerves_system_rpi, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves_system_rpi", update it with "mix deps.update nerves_system_rpi" or clean it with "mix deps.clean nerves_system_rpi"
** (exit) exited in: GenServer.call(#PID<0.224.0>, {:get, "https://github.com/nerves-project/nerves_system_rpi/releases/download/v0.9.1/nerves_system_rpi-v0.9.1.tar.gz"}, 120000)
    ** (EXIT) time out
    (elixir) lib/gen_server.ex:737: GenServer.call/3
    (nerves) lib/nerves/package/providers/http.ex:50: Nerves.Package.Providers.HTTP.download/2
    (nerves) lib/nerves/package/providers/http.ex:20: Nerves.Package.Providers.HTTP.artifact/3
    (elixir) lib/enum.ex:1755: Enum."-reduce/3-lists^foldl/2-0-"/3
    (nerves) lib/nerves/package.ex:99: Nerves.Package.artifact/2
    (mix) lib/mix/task.ex:294: Mix.Task.run_task/3
    (elixir) lib/enum.ex:1229: Enum."-map/2-lists^map/1-0-"/2
    (elixir) lib/enum.ex:1229: Enum."-map/2-lists^map/1-0-"/2
```

### Expected behavior

`mix deps.compile nerves_system_rpi` will compile with any internet connection speed

pinya, 2017-01-25:
Ough, I see it's already fixed in c351cb69e96c17a56ddf044522e96f4da0de089d
Hope 5 minutes will be enough

closed 2017-01-25 by pinya

mobileoverlord, 2017-01-25:
I put a fix in master which should wait infinitely unless the connection is idle for > 2 minutes. If you change your Nerves dependency to point to mater you should be good. I'll be pushing those changes into a release probably today. 👍