nerves-project/nerves #73: Stock mix nerves.new task presents warning about applications/included_applications

ghitchens, 2017-02-13:
Every nerves newbie has to get over the following error message when building firmware, which appears to indicate something is wrong and needs to be fixed.

I'm not sure the proper fix to this.  The builds work regardless, but seems like there should be a way to not get this warning.

```
targhee:testpi3 garth$ mix firmware
[Nerves] Building OTP Release...
==> Building release testpi3:0.1.0 using environment dev
==> One or more direct or transitive dependencies are missing from
    :applications or :included_applications, they will not be included
    in the release:

    :nerves
    :nerves_system_rpi3

    This can cause your application to fail at runtime. If you are sure
    that this is not an issue, you may ignore this warning.
```

fhunleth, 2017-02-13:
The fix is to add `runtime: false` to these dependencies. I started adding this to several projects a week or so ago, but at that time distillery didn't officially support it so it didn't do anything. It looks like a release was just made with the fix, though: https://github.com/bitwalker/distillery/blob/master/CHANGELOG.md#111. If you update to Distillery 1.1.1, you can probably try adding `runtime: false` and see if it fixes it.

GregMefford, 2017-03-02:
Just wanted to let you know that we're making progress on this, and should be able to include it in the next release (which is planned for the end of this week).

We've decided that `nerves` and `nerves_system_*` will be `runtime: false`, but the brand-new `nerves_runtime` package will remain active at runtime, to support the things we actually do want available at runtime without the additional clutter of the build-time tooling and artifact references.

mobileoverlord, 2017-03-07:
Fixed in v0.5.0

closed 2017-03-07 by mobileoverlord