nerves-project/nerves #86: Discuss erlang.cookie error when trying to write to read only FS

mobileoverlord, 2017-03-04:
On the FAQ's we should add a blurb discussing what the obscure error received when Erlang tries to write an erlang.cookie file on the RO FS.

**How to reproduce**
unset `-name` and `-sname` and `-setcookie` in vm.args
Then, boot into the vm and try to start distributed erlang.
```
:os.cmd 'epmd -daemon'
:net_kernel.start([:"my_app@0.0.0.0"])
```

**Solution**
Ensure that a cookie is set in the vm.args even if the vm is starting in no distribution.

GregMefford, 2017-03-06:
We should also use this as an opportunity to warn that the `-extra` option consumes the rest of the options in an unintuitive way, as discovered by @electricshaman.

For example, the following won't work:

```
## Start the Elixir shell
-noshell
-user Elixir.IEx.CLI
-extra --no-halt
-name dev@192.168.0.17
-setcookie somethingsecret
```

Instead, it has to be:
```
## Start the Elixir shell
-noshell
-user Elixir.IEx.CLI
-name dev@192.168.0.17
-setcookie somethingsecret
-extra --no-halt
```

I'm not sure whether this behavior of `-extra` is a bug or by design.

fhunleth, 2017-03-06:
I was hoping that this commit to the new project template would help prevent argument ordering mistake with `-extra`: https://github.com/nerves-project/nerves/commit/a3d09db9c9623d54918371b19b9a5cd3d674865e

Maybe it needs to be more obnoxious like "whatever you do, don't add anything after this" :)

That change may not have been released yet, so maybe @electricshaman didn't have the comment in his vm.args.

mobileoverlord, 2017-03-07:
We can add this to Nerves bootstrap 0.3.1

electricshaman, 2017-03-07:
@fhunleth Definitely didn't see the comment!

GregMefford, 2017-03-07:
To be clear, @electricshaman figured out what the problem was, but it was someone in the Slack channel who actually ran into the problem "in real life." I agree that a prominent comment in the template would be a good idea.

@fhunleth I think the comment in the PR you linked above isn't clear unless you already know about the problem, so it's a good reminder to you but not obvious to someone else. When it says "add custom options here," it's easy to mistakenly read that as "here are some custom options; you should add yours to the end of the list."

I'd propose something at this level of obnoxious, to make it dead simple to do the right thing:

```

## Start the Elixir shell
-noshell
-user Elixir.IEx.CLI

## Set custom options here
#-name nerves@10.0.0.10
#-sname nerves
#-setcookie preferrably_a_random_nonce

## Options added after -extra are interpreted as plain arguments and
## can be retrieved using :init.get_plain_arguments().
-extra --no-halt

```