nerves-project/nerves #87: Documentation: Getting Started section mix firmware generates error message about missing release config file

inactivist, 2017-03-05:
### Environment
* Erlang version: 19
* Elixir version (elixir -v): 1.3.4
* Operating system: Ubuntu Linux 15.10 x64 with Kernel 4.4.8-040408-generic
* Nerves Environment Info:
```
$ mix nerves.env --info
==> distillery
Compiling 18 files (.ex)
Generated distillery app
==> nerves
Compiling 25 files (.ex)
Generated nerves app
==> nerves_system_br
Generated nerves_system_br app
==> nerves_toolchain_ctng
Compiling 2 files (.ex)
Generated nerves_toolchain_ctng app
==> nerves_toolchain_arm_unknown_linux_gnueabihf
Compiling 1 file (.ex)
Generated nerves_toolchain_arm_unknown_linux_gnueabihf app
==> nerves_system_rpi2
Generated nerves_system_rpi2 app
==> hello_nerves
---------------------------
Nerves Environment Packages
---------------------------

Pkg:      nerves_system_br
Vsn:      0.9.4
Type:     system_platform
Provider: [{Nerves.Package.Providers.HTTP, []}, {Nerves.Package.Providers.Local, []}]

Pkg:      nerves_system_rpi2
Vsn:      0.10.0
Type:     system
Provider: [{Nerves.Package.Providers.HTTP, []}, {Nerves.Package.Providers.Local, []}]

Pkg:      nerves_toolchain_arm_unknown_linux_gnueabihf
Vsn:      0.9.0
Type:     toolchain
Provider: [{Nerves.Package.Providers.HTTP, []}, {Nerves.Package.Providers.HTTP, []}]

Pkg:      nerves_toolchain_ctng
Vsn:      0.8.0
Type:     toolchain_platform
Provider: [{Nerves.Package.Providers.HTTP, []}, {Nerves.Package.Providers.Local, []}]

Nerves Loadpaths Start
Nerves Env Start
Nerves Env End
----------------------------
Nerves Environment Variables
----------------------------
target:     rpi2
toolchain:  /home/inactivist/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-0.9.0.linux-x86_64
system:     /home/inactivist/.nerves/artifacts/nerves_system_rpi2-0.10.0.arm_unknown_linux_gnueabihf
app:        /home/inactivist/m/devel/mine/nerves-tests/hello_nerves

Nerves Loadpaths End

```
### Current behavior

Following Getting Started instructions should work as described with no error messages.

### Expected behavior

`mix firmware`  generates the following error message:
```
$ mix firmware
** (Mix)   You are missing a release config file. Run  nerves.release.init task first
```

GregMefford, 2017-03-06:
This was resolved in 20c866d along with #80. Thanks again for taking the time to report the issue! 🎈

closed 2017-03-06 by GregMefford

inactivist, 2017-03-06:
Thanks for merging, @GregMefford.  I'll try to plug myself in to the normal doc review process, if possible.