nerves-project/nerves #91: Can't install dependencies.

LTheGreats, 2017-03-07:
### Environment
* Erlang version (erl -eval 'erlang:display(erlang:system_info(otp_release)), halt().'  -noshell):
* Elixir version (elixir -v): 1.4.0
* Operating system: Ubuntu 16.04
* Nerves Environment Info:
  MIX_TARGET:   rpi
  MIX_ENV:      dev
### Current behavior
When I want to install my dependencies this happens:
```
$ mix deps.get
Env
  MIX_TARGET:   rpi
  MIX_ENV:      dev

Failed to fetch record for 'nerves_runtime' from registry (using cache)
<?xml version="1.0" encoding="UTF-8"?>
<Error><Code>AccessDenied</Code><Message>Access Denied</Message><RequestId>0CD6022F4C66038E</RequestId><HostId>H50ep+f7DZmj3WdJf4RGN35emOP2QPod+/CiJB46usSo1TW/u6je8ITTfp82f5Gx5SiqFFPbzuw=</HostId></Error>
** (Mix) No package with name nerves_runtime (from: mix.exs) in registry
```

### Expected behavior
I expected to have my dependencies installed.

mobileoverlord, 2017-03-07:
Hi @LTheGreats , Sorry, you caught us in the middle of trying to release v0.5.0 of Nerves. Long story short, we needed to go through and update all the toolchains and systems to support it so we rolled some parts back from hex.

The good news is that we are in the process of building all the artifacts necessary to get 0.5.0 up and running today. 

I'll let you know when this is fixed.

Thanks and sorry for things being broken.

mobileoverlord, 2017-03-07:
Things are back up and running. You will need to make sure that you have the latest version of `nerves_bootstrap` installed since I just pushed a fix for it. You can do this by running `mix local.nerves`. Then you'll probably want to recreate your project again since there was an issue in the scaffolding the new proj generator put out. Let me know if you continue to have issues and I'll reopen. :bowtie:  ❤️

closed 2017-03-07 by mobileoverlord