nerves-project/nerves #92: OS X Key Chain prompts for access to unrelated item during custom system build.

brucepom, 2017-03-07:
### Environment
* Erlang version: 19
* Elixir version: 1.4.0
* Operating system: OS X 10.11
* Nerves Environment Info:
 I cloned `nerves_system_linkit`, renamed it and customized the Metadata but didn't make any changes to the system. I just followed the steps for [Customizing Your Own Nerves System](https://hexdocs.pm/nerves/systems.html#customizing-your-own-nerves-system).
### Current behavior

When building the custom system. A terminal prompt appears as follows:
```
The Nerves Docker provider needs to create the image.
Proceed?  [Yn]
```

After hitting `y` to allow Docker to create an image, OSX requests access to a keychain item that's unrelated to Nerves or to my Nerves project. I selected "Deny" and the build appears to continue normally. 

### Expected behavior

After hitting `y` to allow Docker to create an image, if I were prompted to allow access to a Keychain item it would be an item related to Nerves or to the current project.

Note: I don't understand the problem well enough to know if this is an issue with Docker, OS X Keychain, or if it's just something I'm not understanding. Also note I was still able to build the system after clicking "Deny". It shouldn't block anyone's work but could be confusing.

[image]

GregMefford, 2017-03-10:
My best guess based on the error message is that there's something in your project or git config that refers to a repository on GitLab. Do you have a GitLab account and is your custom `nerves_system_linkit_wifi_ap` stored there? Maybe it's trying to check something out from there and requesting your password to clone from there from within the Docker container.

brucepom, 2017-03-10:
Thanks GregMefford, yea that does make sense. I was on a project that used Gitlab about a year ago. I can't imagine how that project could have been associated with anything I'm doing today. I also haven't run into this again since. It only happened once. I'd say we could close this, probably just some weird issue specific to my machine. if it's a real problem I'm sure it will crop up again.

closed 2017-03-10 by brucepom