nerves-project/nerves #108: Format `/root` partition as EXT4

GregMefford, 2017-04-06:
As discussed in #30, we would prefer for the `/root` partition to be EXT4, which will require coordinated changes in several places, including "reformat-on-corruption-or-first-use" in `nerves_runtime` and some changes to `fwup`.

ghitchens, 2017-04-06:
Agreed.   We use EXT4 for /root in all our projects and can't imagine doing otherwise.

ConnorRigby, 2017-04-11:
I have nothing helpful to say here, but i agree `/root` should be ext4. 

as for implementation, I do a check at bootup for a file called `.formatted` on `/root` and if it doesnt exist, i format the block device, then mount it.

ghitchens, 2017-04-13:
@GregMefford I'm not sure if this requires changes to `fwup`, maybe you meant changes to the fwup.conf files?  I think formatting the partition on first use or corruption is a good idea.   It does require mkfs.ext4 to be built in the system.      

@ConnorRigby We don't use a .formatted file, we just assume if the mount fails, the partition needs to be formatted.   Seems like you might have a reason that didn't work for you, wondering if we might be missing something.

ConnorRigby, 2017-04-13:
@ghitchens I actually have erlinit mount it for me. Which fails the first time, since theres no fs on the partition. Im sure theres other ways to do it just the same.

ghitchens, 2017-04-13:
@ConnorRigby oh, ok, I see -- we mount in our elixir startup code so we reformat if the mount fails.

ConnorRigby, 2017-06-14:
I worked with @fhunleth last week and put together a little demo of how i set this up.

* [nerves app](https://github.com/ConnorRigby/hello_nerves_filesystem_demo) - nothing special, just glue for dependencies.
* [nerves_runtime](https://github.com/ConnorRigby/nerves_runtime) - defines behaviors for init systems. (including filesystem) This was ripped from an unpublished project i started. 
* [nerves_system_rpi3](https://github.com/ConnorRigby/nerves_system_rpi3) - has target specific code for handling filesystems (and other init stuff if you so choose)

This is the sort of system i use FBOS and also a couple unpublished projects. the drawback is that theres becomes a lot of duplication inside of `nerves_system_xyz` repositories. I can think of having a common dependency `nerves_system_common`, or even just putting it into `nerves_runtime` 

another idea i toyed with was having `nerves_runtime` define a protocol. and each system define a system struct and then a implementation.

so in `nerves_runtime`
``` elixir
defprotocol Nerves.Runtime.Init do
   def run(opts)
end 

defimp Nerves.Runtime.Init, for: Nerves.System.Common do
   def run(_) do
     # common filesystem init stuff
   end
end
```

then  code that calls that via dynamically building a struct/tagged type.

then inside the `nerevs_system_xyz` repos we would have 

``` elixir
defmodule Nerves.System.Rpi3 do
   defstruct []

   defimpl Nerves.Runtime.Init, for: __MODULE__ do
       @derive [Nerves.System.Common]
   end
end
```

I've never Actually implemented that, but its pretty dry and nice looking code.

fhunleth, 2017-06-15:
I'm pursuing a slightly different approach, but reusing some of your code. The route I'm taking is to add firmware metadata to the installation. This lets you ask a running system what firmware is installed, where it is located (i.e. /dev/mmcblk0), and most importantly, where the application partition is located (i.e. /dev/mmcblk0p3). `nerves_runtime` will be started by `bootloader`. When it starts, if the firmware metadata exists, it will get the application partition, mount it, or format it and mount it. All configuration is stored in the `fwup.conf`. I'm adding this to `nerves_system_rpi0` to try it out.

fhunleth, 2017-06-30:
FYI: This is implemented and being used in `nerves_runtime` and `nerves_system_rpi0` now. I've had a really good experience with it so far.

mobileoverlord, 2017-07-25:
This has rolled out to following supported systems
https://github.com/nerves-project/nerves_system_rpi3/releases/tag/v0.13.0
https://github.com/nerves-project/nerves_system_rpi2/releases/tag/v0.13.0
https://github.com/nerves-project/nerves_system_rpi/releases/tag/v0.13.0
https://github.com/nerves-project/nerves_system_rpi0/releases/tag/v0.15.0
https://github.com/nerves-project/nerves_system_bbb/releases/tag/v0.13.0
https://github.com/nerves-project/nerves_system_linkit/releases/tag/v0.12.0

closed 2017-07-25 by mobileoverlord