nerves-project/nerves #109: Replace automatic nerves_system_ dep naming with multiple function heads

fhunleth, 2017-04-11:
The currently generated `mix.exs` pulls in the system dependency as follows:
```elixir
{:"nerves_system_#{target}", "~> 0.10", runtime: false},
```
This has a few problems: 
1. The version numbers have to be roughly the same between systems for it to work.
2. It obfuscates which system is used to new users
3. It couples `MIX_TARGET` to the system. (e.g., if this were the intent, then why not specify the `MIX_TARGET=nerves_system_rpi3`, etc.)

It seems like an improvement to have `nerves.new` generate a template with an explicit system dep in the `mix.exs`. This would encourage users to create a separate `dep` function for each target. Some of us are doing this already due to the version issue, but I also think that it would improve understanding by decoupling `MIX_TARGET` more from the system image and having an obvious system image reference to see.

Thoughts?

GregMefford, 2017-04-11:
This begs the question of whether we should bring back the `--target` option to `mix nerves.new` so that this one dependency could be generated for the system they intend to use. In that case, would we simply depend on version `>= 0.0.0`?

ConnorRigby, 2017-04-11:
I prefer not having `--target` flag. or make it optional at least. 
We could have tooling for adding a system. 
```
mix nerves.new blinky
mix nerves.system add "nerves_system_rpi3"
```

fhunleth, 2017-04-11:
I'm not sure about `nerves.system add` since that feels like the user should be able to run it after they've substantially edited their mix.exs, and then we'd have to be careful about editing their existing deps. 

Fwiw, I never had bad feelings towards the `--target` option on `nerves.new`. It seemed like a nice convenience for getting started. If it's optional, it would be nice to have a comment with what to fill in for the deps (maybe assume nerves_system_rpi3?) so that if people forget to run it, they can just uncomment some lines and go. Would that be ok?

GregMefford, 2017-04-11:
I think it's reasonable for it to be optional and just default to putting the `rpi3` dep in there by default.

Then again, we could just generate a list of all the popular options so the option is still not needed and it's easy to delete or ignore the ones you don't plan to use. The problem remains that we won't know what version of the system dep to include based on the `--target` option, so at least if they're all there, we can maintain a sane version baseline with each release of `nerves_bootstrap`.

fhunleth, 2017-04-11:
I don't understand your comment on versions. If we're going to generate a nerves system dependency with nerves.new, we have to specify a version, but we do that now. I don't think this change makes versioning any worse in that regard.

GregMefford, 2017-04-11:
We currently assume all the systems are (approximately) version `0.11.0`. I thought part of the intent of this change was to decouple the versioning of the systems from one another, so that we don't have to assume that's the case.

If someone says `--target rpi3` vs. `--target ev3`, we would need to know what version to generate for that dep, unless we just assume that `>= 0.0.0` is good enough for all of them and people can choose a specific version if they want to.

fhunleth, 2017-04-11:
Got it. That assumption that all systems are approximately version `0.11.0` is wrong. For the longest time, the BBB was a version ahead of the other systems. Now the alix, galileo, and ag150 systems are pretty far behind. The constraint to keep all systems roughly the same version is really hard to maintain. I have tried to keep versions the same since I do frequently do directory comparisons between systems to catch differences stylistic and other deltas. However, if I had appreciated the assumption that system version numbers had to be changed together, I would have pushed back sooner.

`>= 0.0.0` isn't really right, but I don't have any better suggestions and I still like the idea that the system is automatically added in nerves.new.

closed 2017-05-30 by mobileoverlord