nerves-project/nerves #112: Dockerfile fails running locale-gen stage: /bin/sh: 1: locale-gen: not found

cthree, 2017-04-14:
The Dockerfile template fails to build:
```
erik@sparrow:nerves_system_rpi0 (master=) $ mix 
==> distillery
Compiling 18 files (.ex)
Generated distillery app
==> nerves
Compiling 25 files (.ex)
Generated nerves app
==> nerves_system_br
Generated nerves_system_br app
==> nerves_toolchain_ctng
Compiling 2 files (.ex)
Generated nerves_toolchain_ctng app
==> nerves_toolchain_armv6_rpi_linux_gnueabi
Compiling 1 file (.ex)
Generated nerves_toolchain_armv6_rpi_linux_gnueabi app
==> nerves_system_rpi0
Generated nerves_system_rpi0 app
|Elixir.Nerves.Package.Providers.HTTP| Downloading Artifact nerves_system_rpi0-0.12.0-dev.armv6_rpi_linux_musl.tar.gz
  From Location:
    https://github.com/tmecklem/nerves_system_rpi0/releases/download/v0.12.0-dev/nerves_system_rpi0-v0.12.0-dev.tar.gz

|Elixir.Nerves.Package.Providers.HTTP| No Available Locations

|Elixir.Nerves.Package.Providers.Docker| Create Image

The Nerves Docker provider needs to create the image.
Proceed?  [Yn] 
Sending build context to Docker daemon 3.584 kB
Step 1/10 : FROM ubuntu:xenial
 ---> 6a2f32de169d
Step 2/10 : MAINTAINER Justin Schneck
 ---> Using cache
 ---> 9d5dadcbd004
Step 3/10 : ENV NERVES_BR_DL_DIR /nerves/cache/buildroot
 ---> Using cache
 ---> fe23d0f005e0
Step 4/10 : ENV LANG en_US.UTF-8
 ---> Using cache
 ---> 2795082da495
Step 5/10 : ENV LANGUAGE en_US:en
 ---> Using cache
 ---> f95777e4edb0
Step 6/10 : ENV LC_ALL en_US.UTF-8
 ---> Using cache
 ---> b86b9624a355
Step 7/10 : RUN locale-gen en_US.UTF-8
 ---> Running in 5c45f6ee4b58
/bin/sh: 1: locale-gen: not found
The command '/bin/sh -c locale-gen en_US.UTF-8' returned a non-zero code: 127
** (Mix) Nerves Docker provider could not create docker volume nerves_cache
```

locale-gen is no longer part of the `ubuntu:xenial` image. Commenting out [Dockerfile line 16](https://github.com/nerves-project/nerves/blob/master/template/Dockerfile#L16) bypasses the issue and the build continues as normal.

Info:
Erlang/OTP 19 [erts-8.3] [source] [64-bit] [smp:8:8] [async-threads:10] [hipe] [kernel-poll:false] [dtrace]
Elixir 1.4.2
macOS
docker Version 17.03.1-ce-mac5 (16048)
rpi* targets

justincjohnson, 2017-04-21:
In case the locale-gen command is important, I believe this can also be fixed by running `apt-get update && apt-get -y install locales` before running `locale-gen en_US.UTF-8`.

mobileoverlord, 2017-04-22:
Thanks for reporting this, I'll update the dockerfile and push a new container to hub

mobileoverlord, 2017-04-22:
I pushed this to docker hub, this should be fixed

closed 2017-04-22 by mobileoverlord