nerves-project/nerves #114: mix.firmware fails when there is a '+' sign in directory name

maxmouchet, 2017-04-19:
### Environment
* Erlang version (erl -eval 'erlang:display(erlang:system_info(otp_release)), halt().'  -noshell): `19`
* Elixir version (elixir -v): `1.4.2`
* Operating system: `macOS 10.12.4`
* Nerves Environment Info:
```
|nerves_bootstrap| Environment Variable List
  target:     rpi3
  toolchain:  /Users/maxmouchet/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-0.10.0.darwin-x86_64
  system:     /Users/maxmouchet/.nerves/artifacts/nerves_system_rpi3-0.11.0.arm_unknown_linux_gnueabihf
  app:        /private/tmp/test +/hello_nerves
```
### Current behavior

```bash
/tmp/test +/hello_nerves % pwd
/tmp/test +/hello_nerves

/tmp/test +/hello_nerves % mix firmware
Env
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Building OTP Release...

==> Cleaning all releases..
==> Clean successful!
usage: dirname path
usage: mkdir [-pv] [-m mode] directory ...
** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, raw: true}
```

### Expected behavior

Should build firmware successfully. It works when there is no '+' sign in the directory name.

fhunleth, 2017-04-20:
Thanks for reporting this. It sounds the arguments to `dirname` and `mkdir` are not being quoted in the `nerves_system_br/create-build.sh` script. I would assume that spaces in directory names breaks as well.

I'll keep this here as a note for one of the core team members (probably me) to fix when we make a pass over nerves_system_br issues.

maxmouchet, 2017-04-20:
Yes I also thought it may be a quoting/escaping problem.

Thanks, have a good day!

closed 2017-04-23 by fhunleth

fhunleth, 2017-04-23:
FYI: The `nerves_system_br` fix needs to propagate to Nerves system images for it to fix this issue. Anyone building systems manually can get it, but users of prebuilt nerves systems will need to wait until the next release of their system.