nerves-project/nerves #120: Create API for garbage collecting global artifacts

mobileoverlord, 2017-05-03:
Keeping the system clean is important. Especially when the artifacts stored globally could easily become version bumped and then just sit there taking up a lot of space.

We should expose some mix commands that let you list / clean the artifacts that are in the global cache.  To assist in this, every time an artifact is used in a project build, we should bump the timestamp on a hidden file in the artifact dir. This will let use display some useful information about when the artifact was last used and possibly even which project used it last.

VinPT, 2019-05-18:
I am just starting learning so I am not that fast however would one mind if I tried my hand at working on this. (first time trying to contribute sorry if I did this wrong)

GregMefford, 2019-06-08:
Hey @VinPT, it would be great if you wanted to give this a try!

The code may be a little complicated to follow what's going on easily, but here is a place to start and maybe experiment locally with getting some artifacts cached in your computer e.g. by building some simple firmware using different targets, and have a look at this code to see how we decide where to store them: https://github.com/nerves-project/nerves/blob/master/lib/nerves/artifact.ex

closed 2021-06-16 by mobileoverlord