nerves-project/nerves #124: fwup version check goes wrong with elixir 1.4.4

melix2002, 2017-05-20:
fwup requirement check in nerves/lib/mix/nerves/utils.ex goes wrong with elixir 1.4.4. Version.requirements() function does not accept version 0.8 as a parameter. Please change it to 0.8.0.

In nerves/lib/mix/nerves/utils.ex 
### Environment
* Erlang version 19.3
* Elixir version 1.4.4:
* Operating system: Linux Mint 18.1

fhunleth, 2017-05-20:
Thanks for the report. We have had reports about strange version check issues with OTP 20 and Elixir 1.4.4, but I never understood what the issue was. It seemed to be resolved by changing OTP versions. For Erlang 19.3, I'm not seeing this. Here's what I have:

```$ mix hex.info
Hex:    0.16.0
Elixir: 1.4.4
OTP:    19.3

Built with: Elixir 1.4.2 and OTP 18.3.4.4
$ iex 
Erlang/OTP 19 [erts-8.3] [source-d5c06c6] [64-bit] [smp:8:8] [async-threads:10] [hipe] [kernel-poll:false]

Interactive Elixir (1.4.4) - press Ctrl+C to exit (type h() ENTER for help)
iex(1)> Version.parse_requirement("~> 0.8")
{:ok, #Version.Requirement<~> 0.8>}
```
Is this what you're saying fails for you?

cc/ @ConnorRigby

melix2002, 2017-05-20:
Thanks for the quick answer. I checked it one more time, and now I know what the exactly issue  is. fwup is returning its version number with an '\n' at the end. So it's even not nerves fault...but fwup's. I saw you have just released 0.14.3 and the issue is still there, so may be it's time for the 0.14.4? :)

`iex(2)> System.cmd("fwup", ["--version"])            
{"0.14.3\n", 0}`
` ** (Version.InvalidVersionError) 0.14.3`
`   (elixir) lib/version.ex:319: Version.to_matchable/2`
`    (elixir) lib/version.ex:165: Version.match?/3`

fhunleth, 2017-05-20:
Ok, that makes more sense. The issue isn't `fwup`, though, since it's almost universal with commandline programs that `--version`'s output is terminated with a newline. It also would make running `--version` manually less user friendly since the version would get combined with a user's shell prompt. Nerves needs to be more robust in its input processing.

Could you add a call to `String.trim/1` in utils.ex to knock off the newline on the version number before it gets used and send a PR? I don't merge PRs on this project, but I'm sure that  @mobileoverlord or @GregMefford will merge it right away knowing that it's the newline that's causing issues.

mobileoverlord, 2017-05-20:
Thanks for submitting this. I am going to push out a patch fix version so more people don't suffer from this problem while moving towards 1.4.4.

fhunleth, 2017-05-21:
Going to close this since it seems resolved. Please reopen if I'm wrong.

closed 2017-05-21 by fhunleth