nerves-project/nerves #130: NERVES_SYSTEM isn't documented

fhunleth, 2017-06-08:
I noticed that neither `NERVES_SYSTEM` nor `NERVES_TOOLCHAIN` were documented or maybe I couldn't find it? This is the main way that I use Nerves, so I was wondering if these should be documented.

electricshaman, 2017-06-28:
I would vote no until we fix the weird discrepancies between NERVES_SYSTEM and the "new way."  You had previously asked me to look into a wrong set/unset message but what I uncovered was a litte more serious than just fixing the message itself

GregMefford, 2017-08-04:
@electricshaman I'm curious if you think the situation has changed at this point, with the updated docs here: https://hexdocs.pm/nerves/systems.html#customizing-your-own-nerves-system

I still want to spend some time with @fhunleth as well, understanding in more detail how his preferred workflow differs, so we can fix this documentation bug.

electricshaman, 2017-08-04:
I'll take a look this weekend!

electricshaman, 2017-08-07:
@GregMefford Looks good!  👍

ConnorRigby, 2018-05-17:
It seems we have docs for this now, but they aren't included in hexdocs. Is there a reason for this?

PR #284 (2018-05-18): Update Internals docs.
This file was added by @mobileoverlord a while back, but never included in the Hexdocs. I added some formatting so it looks a bit better in the tree viewer deal on the left side on hexdocs. Fixes #130
closed 2018-05-18 by mobileoverlord