nerves-project/nerves #141: Don't build Nerves systems when artifact downloads fail

fhunleth, 2017-06-30:
In training meetings, it's not uncommon for someone to have the artifact download fail when building for the first time. It's probably due to us overloading the network. This triggers a Docker build of the system. Since they've never used Nerves before, they don't know that this isn't part of the normal process. When it works, it takes a long time, but usually something isn't installed for the Docker build and the error messages get to be pretty confusing.

I'd really like it if we could download failures error out and let the user fix their network, wait, etc, and try again.

What does everyone think of marking official system releases as "never build"? Obviously, it would be good to have a way to turn this off, but it seems like would be a separate step.

Or, how about making system building "opt-in"? I.e., systems never get locally built until the user enables it in their config.exs. I sort of like this since it seems like we could make the artifact download failure error message say something really helpful and instruct them that building locally is an option. When it does this, the "opt-in" part is the user acknowledging that a really big operation that involves Docket, etc. is ok with them. That way there won't be any surprises.

GregMefford, 2017-07-05:
New users may still follow the instructions in the error message and do the local build since they don't know any better, but I think I like that path better than having it just build it without asking.

mobileoverlord, 2017-07-05:
I think that a better choice would be to isolate the commands that perform network operations from compile operations. 

We should alias `deps.get` so we can perform HTTP provider operations following dependency resolution and output information regarding the download status of artifacts. If there is a problem downloading, you can just continue to call `mix deps.get`. If an artifact for a Nerves package was not downloaded during `deps.get`, it would be built during a call to `mix compile`.

If the operations are properly organized, I don't see a reason to prompt or offer an "opt-in". A system output directory needs to be present at the end of the `deps.precompile` phase otherwise mix needs to be halted to prevent building dependencies incorrectly.

GregMefford, 2017-07-06:
That seems like a good way to go, since downloading the artifact is logically a part of the deps-getting process. There's still a chance that the download fails during `dep.get`, in which case we should have a prominent warning or maybe even an error, saying that an artifact URL was given but it failed to download. If they know what they're doing and intend to build locally, there shouldn't be an artifact URL present at all, right?

mobileoverlord, 2017-07-06:
Right, and if there isn't one present at all `deps.get` can warn them that those artifacts are going to be built when calling `compile`

mobileoverlord, 2018-01-24:
Fixed in v0.9.0
Starting in v0.9.0 artifacts are downloaded after `mix deps.get`

closed 2018-01-24 by mobileoverlord