nerves-project/nerves #158: mix firmware.burn TTY/PTY issue

orestis, 2017-07-31:
### Environment
* Erlang version (erl -eval 'erlang:display(erlang:system_info(otp_release)), halt().'  -noshell): 20
* Elixir version (elixir -v): 1.4.5
* Operating system: Ubuntu Server 16.04
* Nerves Environment Info:
```
Env
  MIX_TARGET:   kiosk_system_rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Environment Package List

  Pkg:      kiosk_system_rpi3
  Vsn:      0.11.0
  Type:     system
  Provider: [{Nerves.Package.Providers.HTTP, []}, {Nerves.Package.Providers.Local, []}]

  Pkg:      nerves_system_br
  Vsn:      0.13.2
  Type:     system_platform
  Provider: []

  Pkg:      nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:      0.10.0
  Type:     toolchain
  Provider: [{Nerves.Package.Providers.HTTP, []}, {Nerves.Package.Providers.HTTP, []}]

  Pkg:      nerves_toolchain_ctng
  Vsn:      0.9.0
  Type:     toolchain_platform
  Provider: []

|nerves_bootstrap| Loadpaths Start

|nerves_bootstrap| Environment Variable List
  target:     kiosk_system_rpi3
  toolchain:  /home/orestis/dev/kiosk_system_rpi3/build_br/host
  system:     /home/orestis/dev/kiosk_system_rpi3/build_br
  app:        /home/orestis/dev/nerves-examples/hello_phoenix/apps/fw

|nerves_bootstrap| Loadpaths End
```
### Current behavior

I have setup a virtual box environment in order to build a custom system. When finally all is built, trying to run `mix firmware.burn` resulted in this error:

```
$ mix firmware.burn
Env
  MIX_TARGET:   kiosk_system_rpi3
  MIX_ENV:      dev

Discovered devices:
0) 32.0 GiB found at /dev/sda
1) 14.84 GiB found at /dev/sdb
Which device do you want to burn to? 1
sudo: no tty present and no askpass program specified
```

After much searching online, I realised the culprit is that when running `System.cmd` there's no pty/tty session attached.

When running via bash:
```
$ tty
/dev/pts/0
```

When running via iex:
```
iex(1)> System.cmd "tty", []
{"not a tty\n", 1}
```

So essentially when running "sudo fwup", sudo complains that there's no TTY attached and cannot ask for a password. A bit more bafflingly, the SUDO_ASKPASS environment variable doesn't do anything, since `System.cmd` doesn't inherit the environment in any case (I think). If I try to force the default SUDO_ASKPASS behaviour by doing `sudo --askpass fwup`, the `x11-ssh-askpass` is indeed invoked and (rightfully) complains there's no display.

### Expected behavior

The `sudo fwup` command should be invoked with the correct PTY passed in so `sudo` can just prompt for a password. 

### Workaround
I had to setup my user to have passwordless sudo in order to keep working.

fhunleth, 2017-07-31:
Yes, this has been a major headache with how Erlang handles console I/O that affects several areas of the Nerves tooling. 

Looping @GregMefford in here so that he's aware of yet another area that this affects.

As for the SUDO_ASKPASS weirdness, that part actually works for me and I believe others which is why this issue probably hasn't been reported before. However, you don't have a display so `sudo` really needs to get a real TTY anyway.

electricshaman, 2017-07-31:
After the nerves 0.7 release, you can skip the need for sudo altogether by giving yourself read/write permissions on the `/dev/sdb` device.  A udev rule for acheiving this might look like:

```
KERNEL=="sdb", SUBSYSTEMS=="block", GROUP="orestis", MODE="0660"
```

It's still an extra step, so it's not ideal but at least it's a narrow solution rather than affecting your whole system like passwordless sudo.

dstreet26, 2017-11-21:
I'm also having this problem. I tried the passwordless sudo trick but that doesn't seem to work in the Windows Linux Subsystem (which up to now has been great).

I'm trying to run

`mix firmware.burn -d out.bin`

With the intention to copy the file onto the SD card with Win32DiskImager

You would think that writing to a local file like `out.bin` wouldn't require sudo in the first place..

fhunleth, 2020-10-03:
I think that we've found that fixing the tty/pty in Erlang is insurmountable, so closing. If someone has a great idea on how to un-`setsid` child processes in Erlang to make this work, I have no doubt that the fix will propagate everywhere including here.

closed 2020-10-03 by fhunleth