nerves-project/nerves #161: Nerves system.shell should not call clean on initialization

mobileoverlord, 2017-08-02:
`mix nerves.system.shell` Upon initialization calls make clean which prevents introspection into the build. Make clean should be called only by nerves.clean

GregMefford, 2017-08-02:
For posterity, the decision to do it this way was that people often don't (and potentially shouldn't) understand at first that Buildroot doesn't handle incremental builds properly, so it's easy to get into a state where the build completes "successfully" but hasn't actually built what you wanted, so it doesn't work and it's hard to figure out why.

We had talked about having the build do a `make clean` by default unless an environment variable was specified so that people familiar with Buildroot could skip that step if they wanted to. To be honest, the reason this variable isn't there now is that I forgot to go back and add it.

Here are a few paths forward as I see them:

1. As this issue suggests, never `make clean` unless the user asks for it explicitly via `make clean`. This would give the most flexibility but also requires that a newcomer to custom Nerves systems would need to know that this step is required to get a reliable build. Maybe that's fine, since they're going to have to read a guide anyway to figure out the `make menuconfig`/`make savedefconfig` dance.

2. Don't `make clean` when running `mix nerves.system.shell`, but always run it from the `mix compile`/`mix firmware` workflow. This gives you a change to go do whatever you want and have full control from within the system configuration shell (without a surprise `make clean`), but when you're done, you always get a clean build. Along with this option, we could still have a `NERVES_SYSTEM_BUILD_INCREMENTAL` variable that, if set, would cause `mix compile`/`mix firmware` to skip the `make clean` step.

electricshaman, 2017-08-02:
I am strongly in the camp of #1.

electricshaman, 2017-08-02:
Option one is also consistent with how Nerves takes a view of things when using the Local provider.  https://github.com/nerves-project/nerves/pull/148

GregMefford, 2017-08-02:
I'm on board with option 1 as well, and it's the simplest to implement (deleting one line from each provider). I'll make a PR to do that for now, since option 2 is actually still part of option 1 and could be implemented separately if desired.

fhunleth, 2017-08-02:
I'll say that I'm in the camp of consistent builds. Here's some good information to read for anyone wanting to do incremental Buildroot rebuilds: http://nightly.buildroot.org/manual.html#full-rebuild

I'll add another option to the discussion and that is to prompt the user before doing a `make clean`. I don't mind if the prompt can be turned off, but I really would like the default to be that people get good builds. My experience has been that the time lost to dirty Buildroot builds far exceeds the Buildroot rebuild time (unless your project brings in webkit or chromium). However, I totally appreciate the pain of throwing out a 30 minute build accidentally. Having an opportunity to CTRL+C out at a prompt if I've forgotten to turn off any autocleaning would be nice.

electricshaman, 2017-08-02:
Yes, I should clarify that I've been building a system that uses webkit a lot lately so my perspective is somewhat skewed.  I timed it at 52 minutes last night.

fhunleth, 2017-08-02:
I feel your pain. Building Chromium feels like building qt-webkit-kiosk projects twice. I really like so many things about Buildroot, but browsers have a way of sucking all enjoyment out of it. 

I think that the following sections of the Buildroot manual are interesting for any thoughts about speeding up this process:

http://nightly.buildroot.org/manual.html#faq-no-binary-packages
http://nightly.buildroot.org/manual.html#faq-speeding-up-build

GregMefford, 2017-08-04:
I've renamed this issue since #163 solved the issue as it was initially titled, but I feel like we still have an open discussion here that doesn't make sense to move to a separate issue as opposed to just continuing it here.

GregMefford, 2018-11-21:
I'm reviewing old issues and I think we could just close this one.

We have the means to clean a broken Buildroot environment via `mix nerves.clean --all` and I think probably most people who are customizing their systems are running `make menuconfig` and friends in the shell, so `make clean` in that same shell would be a pretty easy step to take manually if things aren't working.

mobileoverlord, 2020-03-23:
This is fixed in master.The Nerves build runner will only clean on `mix nerves.clean <package>` or by cleaning all with `mix nerves.clean --all`

closed 2020-03-23 by mobileoverlord