nerves-project/nerves #162: allow mix firmware to pass args through to fwup

mobileoverlord, 2017-08-03:
`mix firmware` and `mix firmware.burn` should allow additional args to be passed through to fwup. This is going to be useful for signing firmware as part of the build of a project. We could support this by either passing "un-captured" args through to fwup
for example:
`mix firmware --sign --private-key 12345`

or by capturing something like `fwup_args` and passing its value through
For Example:
`mix firmware --fwup-args "--sign --private-key 12345"`

This currently goes through the script `rel2fw.sh` which makes the call to fwup so there could be a bit of work here. I will reference a separate issue which tracks the progress on converting this part of the process to Elixir.

fhunleth, 2017-08-03:
The `rel2fw.sh` script should only used to create archives. We definitely should add support to it so that archives can be signed as they're created. The `--sign` option isn't needed for those. I.e., the examples could be:

`mix firmware --fwup-args "--private-key 12345"`
`mix firmware --fwup-args "--private-key-file foo.priv"`
or
`mix firmware --private-key 12345`
`mix firmware --private-key-file foo.priv`

Not that it wouldn't be useful to have a generic way of sending fwup parameters through rel2fw.sh, but I slightly prefer the `--private-key*` options directly on `mix firmware` since that's less typing and makes it look like firmware signing is more of a first class operation.

One other point: I'm not sure that a separate `mix` command for signing firmware is needed. It seems weird, since you could more easily call `mix firmware` and it's not like you can specify an arbitrary `.fw` file. It seems easier to just tell people to run `fwup` if they have a `.fw` file sitting around:

`fwup --sign --private-key-file fwup-key.priv -i myunsignedfirmware.fw -o mysignedfirmware.fw`

GregMefford, 2017-10-07:
I prefer specifying the API of `mix firmware` rather than leaking the fact that these are really just `fwup-args`:

✅ `mix firmware --private-key 12345`
✅ `mix firmware --private-key-file foo.priv`

Would `--private-key*` _always_ imply `--sign`, or could there be other operations that would require a private key? One thing that comes to mind is encrypted firmware, but in that case, you'd probably encrypt it using the recipient's _public_ key or a pre-shared symmetric key, so that would be a different set of parameters.

fhunleth, 2018-04-20:
Everyone fine with closing this issue? The problem seems to be signing firmware and I think everyone signing their `.fw` files does it after running `mix firmware` by calling `fwup` directly.

GregMefford, 2018-04-23:
Seems fine to me.

closed 2018-04-23 by fhunleth