nerves-project/nerves #183: Add a Mix task to return the full path to the firmware file

GregMefford, 2017-08-16:
The goal is to eliminate the need for things like this: https://github.com/fhunleth/nerves_firmware_ssh/blob/master/upload.sh#L25

Possible solutions include:
* New flags to `mix firmware` (e.g. `mix firmware --quiet --path`) that would ensure the firmware file exists and return its full path
* New Mix task (either in `bootstrap` or `nerves`) that would simply return where the firmware _should_ be, but not actually build it (e.g. `mix nerves.info --firmware-path`)

GregMefford, 2018-08-25:
@ConnorRigby Since you closed #283, should we close this too, or should we still come up with a solution? I realize I created this issue, but I think it was based on a Slack discussion we were having where we were thinking it would be nice to have Mix do this kind of stuff for us.

ConnorRigby, 2019-02-20:
Missed this somehow. I think this is safe to be closed. Anyone feel free to reopen if this is needed.

closed 2019-02-20 by ConnorRigby