nerves-project/nerves #203: Retry partial downloads

fhunleth, 2017-09-23:
A few users have reported some confusion with the toolchain tarball fails to expand due to "Unexpected EOFs in archive". This is likely due to an interrupted download. At least, that's when I've seen it. Here's a snippet from @cjfreeze:
```
Chris@Hexa:~/.nerves/artifacts/nerves_system_rpi3-0.16.1.arm_unknown_linux_gnueabihf$ make
umask 0022 && make -C /mnt/c/Users/Chris/Projects/Elixir/local_pi/local_pi_fw/deps/rpi3/nerves_system_br/buildroot-2017.05 O=/home/Chris/.nerves/artifacts/nerves_system_rpi3-0.16.1.arm_unknown_linux_gnueabihf/.
>>> toolchain-external-custom  Extracting
xzcat /mnt/c/Users/Chris/Projects/Elixir/local_pi/local_pi_fw/deps/rpi3/nerves_system_br/buildroot-2017.05/dl/nerves_toolchain_arm_unknown_linux_gnueabihf-0.11.0.linux-x86_64.tar.xz | tar --strip-components=1 -C /home/Chris/.nerves/artifacts/nerves_system_rpi3-0.16.1.arm_unknown_linux_gnueabihf/build/toolchain-external-custom --exclude='usr/lib/locale/*'   -xf -
xzcat: /mnt/c/Users/Chris/Projects/Elixir/local_pi/local_pi_fw/deps/rpi3/nerves_system_br/buildroot-2017.05/dl/nerves_toolchain_arm_unknown_linux_gnueabihf-0.11.0.linux-x86_64.tar.xz: Unexpected end of input
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
make[1]: *** [/home/Chris/.nerves/artifacts/nerves_system_rpi3-0.16.1.arm_unknown_linux_gnueabihf/build/toolchain-external-custom/.stamp_extracted] Error 2
make: *** [_all] Error 2
```
One possible implementation would be to verify SHA256's of the downloaded tarballs in a similar manner to Buildroot.

mobileoverlord, 2018-01-08:
I’ve been working through this issue and was curious specifically about the title. It states that we should retry downloading the artifact if the download fails. I could see how in the current workflow that this might be desirable considering that in most circumstances Nerves env will attempt to build the artifact if there is an issue with the download.

While working through this issue I felt inclined to investigate the workflow as being part of the problem. For a while we’ve been kicking around the idea that we should separate the network download stage from the build phase. The new proposed workflow will move the network fetch operations to be called as an alias to `deps.get`. In this workflow, I believe that we should not automatically retry but instead only render a clear error / success status in addition to which Nerves packages are marked for build in the build phase.

This will require an update to the aliases that are set in projects to include `”deps.get”: [“deps.get”, “nerves.deps.get”]`
I was going to investigate the possibility of being able to ask Mix for the defined aliases and if it is missing an alias for Nerves in this phase we present a update warning and call the HTTP provider during build time like we are today.

fhunleth, 2018-01-08:
Fwiw, I didn't intend the retry part to be automatic. I just wanted it work work that if the download failed that you wouldn't get a stale partially downloaded file that would error out whenever you try to build again. 

Also, I think this issue could be updated depending on which way that you wanted to go. The example above concerns `make` in a system shell, but download failures outside of the system shell are probably more important since they affect new users.

mobileoverlord, 2018-01-24:
Fixed in v0.9.0

closed 2018-01-24 by mobileoverlord