nerves-project/nerves #205: Can't burn an SDCard from WSL

fhunleth, 2017-10-05:
Nerves mostly works under the Windows Subsystem for Linux, but there's a catch-22 when running `fwup`:

1. If `fwup` is installed in WSL, then creating the `.fw` file works fine. However, `mix firmware.burn` fails since `fwup` can't access the MicroSD card. For whatever reason, Windows doesn't expose it inside of WSL.

2. If `fwup` is installed in Windows (not WSL), then burning an SDCard from WSL is possible. This works since `.exe` files outside of WSL are available inside of WSL. However, filename paths are not translated between WSL and Windows. This prevents `fwup` from opening the files passed on the commandline or referenced inside of the `fwup.conf` files. In other words, `fwup.exe` tries to open a path using the Linux way of reference that path. That makes no sense to it and the open fails.

tmecklem, 2017-12-15:
If you've managed to create a .fw file using `mix firmware`, you can use the host Windows fwup (the chocolatey version) to burn the SD card.

Steps:

1. generate a fw within WSL file using `mix firmware` (usually located in `_build/<target>/dev/nerves/images/<project name>.fw`)
2. copy the fw file to the host: `cp <fw file location> /mnt/c/Users/<username>/Desktop`
3. use fwup on the host to burn the SD card. In an Administrative cmd.exe console window, `fwup C:\Users\<username>\Desktop\<fw file name>`. It should detect the SD card and Just Work :tm:

I'm assembling a longer post with more in depth specific instructions. Let me know if you try this and encounter any problems and I'll incorporate it into the post.

mobileoverlord, 2018-02-16:
I believe that there was a story for this recently. @tmecklem could you update / close if there is a method in place for this?

tmecklem, 2018-02-16:
I ended up making a Windows walkthrough video here https://youtu.be/rzV0qfhzzqc. It elaborates on the previous instructions and should suffice in demonstrating the method if the above steps aren't clear enough. 

It was also noted by @michaelkschmidt in Slack: if you create the nerves project within `/mnt/c`, you don't need the intermediate step of copying it over to there for the `mix firmware.burn`, and you can use a Windows file editor to work on the project. All good things!

I suspect there may be a few more gotchas as we improve Windows support, but this seems to be a sufficiently-good-enough workflow to close this issue.

closed 2018-02-16 by tmecklem