nerves-project/nerves #240: Custom systems don't end up in the right place after compiling

GregMefford, 2018-02-01:
### Environment
* Erlang version: `20`
* Elixir version (elixir -v): `1.6.0`
* Operating system: `OSX Sierra`
* Nerves Environment Info:
  (mix nerves.env --info) or
  General info about target / system / toolchain

```
  Pkg:      nerves_system_br
  Vsn:      0.17.0
  Type:     system_platform
  Provider: nil

  Pkg:      nerves_system_rpi0_zbar
  Vsn:      0.1.0
  Type:     system
  Provider: {Nerves.Artifact.Providers.Docker, []}

  Pkg:      nerves_toolchain_armv6_rpi_linux_gnueabi
  Vsn:      0.13.0
  Type:     toolchain
  Provider: {Nerves.Artifact.Providers.Local, []}

  Pkg:      nerves_toolchain_ctng
  Vsn:      1.3.0
  Type:     toolchain_platform
  Provider: nil
```

### Current behavior

```
tar: Error opening archive: Failed to open '/Users/greg/.nerves/dl/nerves_system_rpi0_zbar-portable-0.1.0-E5AC4329AA17C09B3B2D5B6BCDC1C09F3FB0E59DC069B99ED4AFCCFC27622A87.tar.gz'
could not compile dependency :nerves_system_rpi0_zbar, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves_system_rpi0_zbar", update it with "mix deps.update nerves_system_rpi0_zbar" or clean it with "mix deps.clean nerves_system_rpi0_zbar"
** (MatchError) no match of right hand side value: {:error, ""}
    lib/nerves/artifact/cache.ex:20: Nerves.Artifact.Cache.put/2
    lib/mix/tasks/compile.nerves_package.ex:25: Mix.Tasks.Compile.NervesPackage.run/1
    (mix) lib/mix/task.ex:314: Mix.Task.run_task/3
    (mix) lib/mix/tasks/compile.all.ex:63: Mix.Tasks.Compile.All.run_compiler/2
    (mix) lib/mix/tasks/compile.all.ex:47: Mix.Tasks.Compile.All.do_compile/4
    (mix) lib/mix/tasks/compile.all.ex:18: anonymous fn/1 in Mix.Tasks.Compile.All.run/1
    (mix) lib/mix/tasks/compile.all.ex:34: Mix.Tasks.Compile.All.with_logger_app/1
    (mix) lib/mix/task.ex:314: Mix.Task.run_task/3
```

### Expected behavior

It should build the firmware using the custom system.

GregMefford, 2018-02-01:
Sorry, I forgot to mention that this compiled the system fine, it just failed to find it when trying to build firmware in the actual project.

mobileoverlord, 2018-02-01:
This should be fixed in branch `rel-v0.9.4` can you give it a test and let me know?

GregMefford, 2018-02-01:
Confirmed - after cleaning out the stale version of the custom system and running again with this branch, it resolved the issue and built a firmware. 👍

mobileoverlord, 2018-02-10:
Fixed in v0.9.4

closed 2018-02-10 by mobileoverlord