nerves-project/nerves #248: Add documentation around requiring `runtime: false` for ALL `:nerves` dependencies

entone, 2018-02-15:
Received this error

`** (Mix) Stopping package build due to errors.
Dependencies excluded from the package (only Hex packages can be dependencies): nerves`

@mobileoverlord somehow knew that I had not marked one of my `:nerves` dependencies as `runtime: false`

We should document that this is required now.

mobileoverlord, 2018-02-16:
The problem is amplified for the `:nerves` dependency since so many depend on it. The real issue here is when calling `mix firmware` distillery doesn't know how to construct a boot script when an app that is the dependency of many children has a mix of declaring `runtime: false`. the dependency option needs to be consistent on all declarations

GregMefford, 2018-11-21:
I'd be happy to write up some docs about this, but I'm struggling to figure out where it would make sense / if it's even necessary, because the generators and official systems are all already doing the right thing at this point, so the only time you'd need to worry about this is when upgrading an old system or project that doesn't follow the convention yet. Maybe there should just be a note in the [upgrade guide](https://github.com/nerves-project/nerves/blob/master/docs/Updating%20Projects.md)?

mobileoverlord, 2020-03-23:
Nerves uses Elixir releases when using Nerves ~> 1.5 and Elixir ~> 1.9. When using this configuration, all the dependency issues will be reconciled by `mix deps.get`.

closed 2020-03-23 by mobileoverlord