nerves-project/nerves #254: `SQUASHFS error: ` on raspberry pi

amol-c, 2018-03-03:
### Environment
* Erlang version (erl -eval 'erlang:display(erlang:system_info(otp_release)), halt().'  -noshell): 20

* Elixir version (elixir -v): Erlang/OTP 20 [erts-9.2] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:10] [hipe] [kernel-poll:false]
Elixir 1.6.1 (compiled with OTP 19)

* Operating system: Mac OS Sierra 10.12.6
* Nerves Environment Info:
  (mix nerves.env --info) or
  General info about target / system / toolchain

|nerves_bootstrap| Environment Package List

  No packages found
|nerves_bootstrap| Loadpaths Start

NERVES_SYSTEM is unset
NERVES_TOOLCHAIN is unset
|nerves_bootstrap| Environment Variable List
  target:     host
  toolchain:  unset
  system:     unset
  app:        $HOME/firmware

|nerves_bootstrap| Loadpaths End

*Build script:
#!/bin/bash
MIX_TARGET=rpi mix deps.get
MIX_TARGET=rpi mix firmware
MIX_TARGET=rpi mix firmware.push 192.168.1.161

### Current behavior
The app runs fine for few minutes/hours on raspberry pi. But eventually it starts throwing the below error:
```
SQUASHFS error: Unable to read directory block 
SQUASHFS error: Unable to read metadata cache directory 
```

### Expected behavior
The app should either restart if there is any system error or should continue to work normally.
OR
Is there a way to catch this kind of system error ?

fhunleth, 2018-03-03:
This is a really good point. I've only ever handled these errors indirectly when they start causing other failures. I looked through the Linux kernel source code and I don't see any squashfs options for this. Other filesystems have "onerror" or "error" options to do things like remount read-only or panic. I don't see anything that propogates to userland. The only possibility that I see so far is to monitor the log for the "SQUASHFS error" string and that's not very satisfying.

amol-c, 2018-03-04:
Thanks for your response. 

My knowledge about working with kernel is quite limited. From whats at [this](https://elinux.org/images/d/d7/Elce2010-flash-filesystems.pdf) link, `However, as it is read­only, it works fine with mtdblock, as long as the flash chip doesn't have any bad blocks`

I think my memory card had bad blocks and the way I fixed it (lazily) was to just replace it with a spare one. But, I do think if there could be a longer term fix around this that would be awesome (like replace with another `FS` that handles bad blocks.). There was a discussion about the issue with squashfs [here](https://bbs.nextthing.co/t/replacing-ubifs-with-squashfs/6825/11) .

From what I have been reading around `squashfs`, it does sound like its pretty unreliable or may be I am completely wrong.

fhunleth, 2018-03-04:
I think that the Internet has misled you with something unrelated. Rest assured SquashFS is quite reliable, in very many shipping products, and I could imagine arguing that it's more reliable for its intended usage. 

You're dealing with either a bad MicroSD card or a flaky MicroSD card holder. Both of those are real things that can happen in production devices, so I like the idea of being able to detect the issue, but I don't know how to handle it. 

Fwiw, in the projects that I've worked on, we've either 1. switched to eMMC, 2. been careful about where we source our MicroSD cards, and/or 3. used MicroSD card holders with latching mechanisms and sometimes gluing them down. We also benefit that in a production device, the MicroSD holder doesn't get exercised after manufacturing, so unless the end user is violent with the device, it shouldn't get much wear. I've had a very good experience with MicroSD cards in production, but for my hobbyist use where I buy cheap no-name cards, I do run into a few failures. They're all been catastrophic, though, so I'm not sure that detection would do me much good.

What you found in your searches relates to raw NAND flash devices. MicroSD cards (and SSDs) contain microcontrollers that handle bad blocks, do wear leveling, and several other things to keep data safe and make the device easy to use. Raw NAND devices don't have any of this and rely on the OS to provide them. If you don't, it's total disaster. 

Somewhat complicating the discussion is that Linux doesn't have a flash translation layer for raw NAND that can replicate what's included in "managed devices" like MicroSD cards, SSDs, and eMMC memory. Linux only has special Flash-aware filesystems like UBIFS. It is possible to have SquashFS on top of UBI (the lower level part of UBIFS) and at least one company working with Nerves has done that to protect the SquashFS filesystem from NAND issues. However, they have a high volume product and can spare substantial engineering resources to re-implement what you get in managed devices. However, you're not dealing with raw NAND flash, so none of this applies to you.

michaelkschmidt, 2018-03-05:
@amol-c, @fhunleth  is correct--SD Cards are "managed" NAND devices and do bad block management transparently in the background.  

Losing the ability to read a block is not a normal use case for NAND.  Every time a page is read, error correction codes are applied to it, and when they reach a certain threshold the page is moved to a "good" block.  Your error indicates this process failed.  UBI won't help in this scenario, as it expects errors to occur gradually.  Like @fhunleth said this all points to bad hardware of some kind

amol-c, 2018-03-05:
That makes sense. Thanks for an illuminating discussion.

🙏

fhunleth, 2018-03-05:
I'm going to close this issue since I don't think that we're going to have a solution any time soon. However, if anyone comes up with anything, please suggest or reopen.

closed 2018-03-05 by fhunleth