nerves-project/nerves #260: ssh in docket container does not exist

mattludwigs, 2018-03-08:
### Environment
* Erlang version: "20"
* Elixir version: 1.6.2
* Operating system: Mac OSX 10.13.3
* Nerves Environment Info:
```
 toolchain:  ~/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-darwin_x86_64-1.0.0-rc.0

Pkg:      nerves_system_br
Vsn:      1.0.0-rc.1
Type:     system_platform
Provider: nil
```
### Current behavior

While in the nerves system shell I try to run ssh I get:

`bash: ssh: command not found`

The use case is cloning a custom private git repo via ssh to build. Stack trace from `mix nerves.system.shell`:

```
Doing shallow clone
Cloning into 'zipgateway-37d9c9e98fb7a9260b1a4bfd675ddfdfdab1f104'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
Shallow clone failed, falling back to doing a full clone
Doing full clone
Cloning into 'zipgateway-37d9c9e98fb7a9260b1a4bfd675ddfdfdab1f104'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
```

### Expected behavior

That ssh exists in the docker container.

This will be useful for cloning and building custom git project, behind a private repo primarily, as part of the build. While there are probably something things in place to manage ssh keys between the host and the container, at least having `ssh` available will be the first step to be able to do this.

mobileoverlord, 2018-03-08:
I think its a good idea to add `openssh-client` to the default image since buildroot is going to expect that its installed for use to fetch from these private repos. The next step is going to be, as you mentioned, figuring out how to share the keys without leaving them behind.

erauer, 2018-03-08:
@mobileoverlord https://dchua.com/2016/01/15/ssh-agent-forwarding-with-your-docker-container/ may be useful. I've used this technique before on a Linux host.

mobileoverlord, 2018-03-09:
@erauer thanks for the pointer. Looks like that will work :)

mobileoverlord, 2018-03-12:
Its starting to look like this issue is deeper.

https://github.com/docker/for-mac/issues/410

I have verified that https://github.com/nerves-project/nerves/pull/262 works on linux, but not on mac OS. I was running docker (edge).

I have see some work arounds by doing a two part dance with socat and others that involve standing up a separate container, but I am worried about leaking keys and access through these methods. Not sure what the best way to go about fixing this is.

closed 2018-03-16 by mobileoverlord