nerves-project/nerves #268: Nerves environment output interferes with mix format

dkarter, 2018-03-26:
### Environment

* Erlang version (20):
* Elixir version (1.6.4 [compiled with OTP 19]):
* Operating system: macOS 10.13.3
* Nerves Environment Info:

```
Nerves environment
  MIX_TARGET:   rpi0
  MIX_ENV:      dev

|nerves_bootstrap| Environment Package List

  Pkg:      nerves_system_br
  Vsn:      0.17.0
  Type:     system_platform
  Provider: nil

  Pkg:      nerves_system_rpi0
  Vsn:      0.21.0
  Type:     system
  Provider: {Nerves.Artifact.Providers.Docker, []}

  Pkg:      nerves_toolchain_armv6_rpi_linux_gnueabi
  Vsn:      0.13.1
  Type:     toolchain
  Provider: {Nerves.Artifact.Providers.Local, []}

  Pkg:      nerves_toolchain_ctng
  Vsn:      1.3.1
  Type:     toolchain_platform
  Provider: nil

|nerves_bootstrap| Loadpaths Start

|nerves_bootstrap| Environment Variable List
  target:     rpi0
  toolchain:  /Users/dkarter/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-darwin_x86_64-0.13.1
  system:     /Users/dkarter/.nerves/artifacts/nerves_system_rpi0-portable-0.21.0
  app:        /Users/dkarter/dev/hardware_projects/open_sesame

|nerves_bootstrap| Loadpaths End
```

### Current behavior

The Nerves framework is hijacking the output of every mix task and injects the Nerves environment text. This prepends text to every mix command, which can be a problem for mix commands in which the user relies on the output of the command (e.g. the code formatter).

[image]

To test run `cat rel/config.exs | mix format -`

The output of this command is what used by some editors to get the formatted output of the file and replace the content, resulting in syntax errors.

Another use case comes to mind is `mix phx.gen.secret`:

[image]

### Expected behavior

Only output of specific tasks such as `mix firmware`, `mix firmware.push` or `mix compile` should be modified instead of all mix tasks.

---

I can see how this can be thought of as a code formatter bug instead, but since it is affecting the usage of all mix tasks, I thought I'd bring it up here first to see if that's something you'd want to fix on Nerves' side.

Thanks!

mobileoverlord, 2018-03-26:
In preparation for the next version of `nerves_bootstrap` you can run `mix archive.install github nerves-project/nerves_bootstrap`

dkarter, 2018-03-26:
Works great! Thank you! 🌟