nerves-project/nerves #270: Rename `provider` terminology to `build_runner`

fhunleth, 2018-04-20:
I'm finding the word `provider` hard to explain to people for what it does until I tell them the options. It selects they location and build environment that Nerves systems are made. The options are:
* `local` - if you're on Linux, you can build systems locally
* `docker` -  if you're not on Linux, this brings up a Docker container
* `ec2` - this is a hypothetical "provider" that makes a request to an EC2 server to build the system for you and once it's done, it downloads the result. We had something like this with Bakeware a couple years ago.

The best idea I have so far is to rename it `build_runner`, but I'm open to others. I'd like to get the rename done before the 1.0.0 release so that the provider nomenclature isn't with us forever. The ramifications of the naming should be minor since only two people (to our knowledge) have created custom providers and they're both capable of dealing with the rename.

tmecklem, 2018-04-20:
Almost as generic as `provider`, it could be `build_strategy`. I don't love that name, but maybe it'll help kickstart some other ideas.

electricshaman, 2018-04-20:
I'm 100% for this.  Provider is way too abstract.  I like `build_runner` because it's pretty self-evident.

mobileoverlord, 2018-04-25:
Since `provider` and `provider_opts` are properties that can be overridden on the `nerves_package` compiler options how should we handle transitioning this to the new namespace? Do we need to provide deprecation or should we just flip the switch and force an upgrade?

fhunleth, 2018-04-25:
I'm fine with failing out if they're specified. I don't think that they're used by that many people, right?