nerves-project/nerves #275: Conditional Erlinit configs.

ConnorRigby, 2018-04-27:
It would be nice to not have to specify different rootfs additions for each situation you want to modify a setting in Erlinit.

it'd be nice to be able to do something like:
```elixir
config: :nerves, firmware: [
   erlinig_config("erlinit.#{target}.config")
]
```

fhunleth, 2018-04-27:
The trick is putting the `erlinit.config` in a deterministic location. That means that it can't be under an OTP application's `priv` directory since then `erlinit` would need to scan all of the apps and try to figure out which one is the desired one. However, many other places would work and if `erlinit` needs to be changed to look in a new location, that's an easy change for me to implement.

ConnorRigby, 2018-04-27:
I think that is one solution, but again i don't know if it is _the_ solution. I remember @electricshaman having some ideas about this also.

mobileoverlord, 2020-03-23:
This can be handled in the mix config directly now. https://hexdocs.pm/nerves/advanced-configuration.html#overriding-erlinit-config-from-mix-config

closed 2020-03-23 by mobileoverlord