nerves-project/nerves #291: mix firmware.burn -d <filename>

critch, 2018-06-22:
### Environment

* Erlang version (20):
* Elixir version (1.6.2):
* Operating system: Debian 8.6
* Nerves Environment Info:
   Pkg:         nerves_system_br
  Vsn:         1.2.2
  Type:        system_platform
  BuildRunner: nil

  Pkg:         nerves_system_orangepi_pcplus
  Vsn:         0.0.52
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.0.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.4.0
  Type:        toolchain_platform
  BuildRunner: nil

### Current behavior
I build on a server in a remote closet. I want to build into a file that I can copy to my laptop and write to a sdcard for booting. If the file does not exist prior to running the mix firmware.burn -d <filename> command, it fails as a writeable location and therefore wants to run askpass. Being as my access is via ssh with no X11, there isn't a workable askpass to use, and the process fails.

### Expected behavior
Closest I can figure out is the first test should be for exists. If it doesn't exist, see if we can touch a file at that path. If so and writeable, then go to writing. If touch fails or isn't writable, then finally fail into askpass. We might want to figure out a way to handle askpass in another way as an SSH only session doesn't have a current askpass, and in this instance, shouldn't be needed.

mobileoverlord, 2018-06-22:
I agree, we should attempt to see if elevated permission is required before executing fwup.

mobileoverlord, 2018-06-22:
related https://github.com/nerves-project/nerves/issues/158

closed 2020-03-23 by mobileoverlord