nerves-project/nerves #293: Are there any plans to support OpenWrt and BuildRoot?

ponyatov, 2018-06-26:
The closest devices I have:
- HLK-RM04 (RalInk RT5350F)
- VoCore (old v.1)
- AT91SAM9 deb board

Is any manual exists how to build BuildRoot and OpenWrt firmwares, to be usable with Nerves/Elixir?

ponyatov, 2018-06-26:
PS: Should work in parallel with custom firmwares, not full init/runtime replacement
(LuCI, routing with GSM and misc external modems, specially written scripts for OEM modules,..)

fhunleth, 2018-06-28:
What you're looking for is the custom Nerves system documentation at https://hexdocs.pm/nerves/systems.html#customizing-your-own-nerves-system. The documentation assumes that you can find an existing Nerves port and then customize it to your needs. We use Buildroot and you'll end up working with the Buildroot config files.

We don't support OpenWrt. With a lot of work, it's possible to pull out the OpenWrt patches and put them in a form that's suitable for Buildroot.

I don't understand your "ps", Nerves replaces /sbin/init. If you don't want that, Elixir can be installed in Buildroot and OpenWRT for a more traditional embedded Linux setup.

closed 2018-06-28 by fhunleth