nerves-project/nerves #302: [Regression] mix.firmware fails when there is a ' ' in directory name

schainker, 2018-07-31:
### Environment

* Erlang version (`erl -eval 'erlang:display(erlang:system_info(otp_release)), halt().'  -noshell`): 21
* Elixir version (`elixir -v`): Elixir 1.6.5 (compiled with OTP 19)
* Operating system: Mac OS High Sierra 10.13.6
* Nerves Environment Info:
  (`mix nerves.env --info`) or
  General info about target / system / toolchain

```|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.3.2
  Type:        system_platform
  BuildRunner: nil

  Pkg:         nerves_system_rpi3
  Vsn:         1.2.1
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.0.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.4.0
  Type:        toolchain_platform
  BuildRunner: nil

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi3
  toolchain:  /Users/me/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-darwin_x86_64-1.0.0
  system:     /Users/me/.nerves/artifacts/nerves_system_rpi3-portable-1.2.1
  app:        /Users/me/stealth+ mode/nerves_examples/hello_leds

|nerves_bootstrap| Loadpaths End
```

### Current behavior

Similar to #114:  If there's a space in the path of the directory you're building from, you'll receive a build error like so: 

```
$ mix firmware

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Building OTP Release...

==> Cleaning all releases..
==> Clean successful!
Updating base firmware image with Erlang release...
Copying rootfs_overlay: /Users/me/stealth+         ##### The path cuts off after the space
rootfs_overlay: mode/nerves_examples/hello_leds/rootfs_overlay does not exist!           ### And this is the rest of the path after the space
** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, raw: true}
```

If there is no whitespace in the path, `mix.firmware` works normally.  We have regressed by change of character, so to speak.

### Expected behavior

`mix.firmware` should build normally if your path has spaces in it.  I assume there are caveats about using windows and whitespace.

schainker, 2018-07-31:
Just to be clear, `+` in the name no longer causes a problem, just whitespace.  Just tested.

mobileoverlord, 2018-07-31:
Thats a bummer. The issue seems to be in how we are enumerating the rootfs overlays in the `rel2fw.sh` script here: https://github.com/nerves-project/nerves_system_br/blob/master/scripts/rel2fw.sh#L126

Since space is the default delimiter, its breaking it into too many parts. If you wanted to fiddle with that to see what you can get working, the location of the script it is referencing is going to be at `/Users/me/.nerves/artifacts/nerves_system_rpi3-portable-1.2.1/scripts/rel2fw.sh`

fhunleth, 2018-07-31:
I think the simplest answer here is to run a shell script linter on our scripts. I think there will be a lot of double quotes that need to be added. Going forward, we should also add a CI check that invokes the linter since it's too easy to mess this up.

fhunleth, 2018-08-02:
Thanks for taking a look into this! I have some fear of changing `$IFS`. This is a pretty mindless task, but I'd like to fix all of the `shellcheck` issues first and then if there's still a problem, then look at `$IFS`. Here's an example:

```sh
 $ shellcheck create-build.sh

In create-build.sh line 29:
        echo "`pwd -P`/$filename"
              ^-- SC2006: Use $(..) instead of legacy `..`.

In create-build.sh line 41:
    BUILD_DIR=o/$(basename -s _defconfig $DEFCONFIG)
                                         ^-- SC2086: Double quote to prevent globbing and word splitting.

In create-build.sh line 45:
mkdir -p $BUILD_DIR
         ^-- SC2086: Double quote to prevent globbing and word splitting.

```

You can probably see that this will be painful, but I think that getting everything in double quotes will fix this. If you do this, could you send a PR for a little bit at a time? Maybe one file or so at a time so that it's not a scary large PR.

schainker, 2018-08-15:
We feel the same about `$IFS` here, for sure.  

👍  with getting shellcheck modifications done first to see if "best practices" clear the way for us.  I'll take a crack at this, slowly but surely.

fhunleth, 2020-10-03:
Closing this since we regularly fix scripts in this area, and it will be easier to track new instances with specific failures. In general, though, spaces in paths have caused issues with third party libraries and `elixir_make` users, so it seems like you can save quite a bit of time by avoiding them.

closed 2020-10-03 by fhunleth