nerves-project/nerves #322: mix firmware fails when using a dependency that depends on elixir_make

prook, 2018-08-22:
### Environment

* Erlang: 21
* Elixir: 1.7.2
* Operating system: Ubuntu 18.04.1
* Nerves Environment Info:
  * nerves: 1.3.0
  * nerves_bootstrap: 1.1.0
  * nerves_toolchain_ctng: 1.5.0
  * nerves_toolchain_arm_unknown_linux_gnueabihf: 1.1.0
  * nerves_system_br: 1.4.1
  * nerves_system_bbb: 1.3.0

### Current behavior

Set up a project with `elixir_make` dependency, and a nerves firmware project depending on that project:

```shell
$ cd `mktemp -d`
$ mix new xyz
$ yes | mix nerves.new fw
```
In `xyz/mix.exs`:
```elixir
  defp deps do
    [
      {:elixir_make, "~> 0.4", runtime: false}
    ]
  end
```
In `fw/mix.exs`:
```elixir
  defp deps(target) do
    [
      {:nerves_runtime, "~> 0.6"},
      {:xyz, path: "../xyz"},
    ] ++ system(target)
  end
```
Next, attempt to build the firmware:
```shell
$ cd fw
$ MIX_TARGET=bbb mix firmware

Nerves environment
  MIX_TARGET:   bbb
  MIX_ENV:      dev

|nerves_bootstrap| Building OTP Release...

==> Release failed, during .boot generation:
        Undefined applications: [elixir_make]
```
Removing `runtime: false` from `elixir_make` dependency in `xyz` allows me to build the firmware, however, `elixir_make` is not a runtime dependency.

Is this a bug or am I missing something?

mobileoverlord, 2018-08-22:
I was unable to reproduce this. Maybe you have a dependency that is out of date? I think there were some updates to `distillery` 2 that address this. Here are my deps
```
* nerves_system_br 1.4.1 (Hex package) (mix)
  locked at 1.4.1 (nerves_system_br) 58a85d4d
  ok
* nerves_system_linter 0.3.0 (Hex package) (mix)
  locked at 0.3.0 (nerves_system_linter) 84e0f63c
  ok
* system_registry 0.8.0 (Hex package) (mix)
  locked at 0.8.0 (system_registry) 09240347
  ok
* elixir_make 0.4.2 (Hex package) (mix)
  locked at 0.4.2 (elixir_make) 332c649d
  ok
* test_dep 0.1.0 (../test_dep) (mix)
  ok
* nerves_runtime 0.6.5 (Hex package) (mix)
  locked at 0.6.5 (nerves_runtime) 7b43e555
  ok
* artificery 0.2.6 (Hex package) (mix)
  locked at 0.2.6 (artificery) f6029097
  ok
* distillery 2.0.4 (Hex package) (mix)
  locked at 2.0.4 (distillery) 7fdc6647
  ok
* nerves 1.3.0 (Hex package) (mix)
  locked at 1.3.0 (nerves) 473b95af
  ok
* nerves_toolchain_ctng 1.5.0 (Hex package) (mix)
  locked at 1.5.0 (nerves_toolchain_ctng) 34b8f566
  ok
* nerves_toolchain_arm_unknown_linux_gnueabihf 1.1.0 (Hex package) (mix)
  locked at 1.1.0 (nerves_toolchain_arm_unknown_linux_gnueabihf) ca466a65
  ok
* nerves_system_bbb 1.3.0 (Hex package) (mix)
  locked at 1.3.0 (nerves_system_bbb) 2cc83426
  ok
* shoehorn 0.4.0 (Hex package) (mix)
  locked at 0.4.0 (shoehorn) f3830e22
  ok
```

prook, 2018-08-22:
I'm at exactly the same versions:

```
* nerves_system_br 1.4.1 (Hex package) (mix)
  locked at 1.4.1 (nerves_system_br) 58a85d4d
  ok
* nerves_system_linter 0.3.0 (Hex package) (mix)
  locked at 0.3.0 (nerves_system_linter) 84e0f63c
  ok
* system_registry 0.8.0 (Hex package) (mix)
  locked at 0.8.0 (system_registry) 09240347
  ok
* elixir_make 0.4.2 (Hex package) (mix)
  locked at 0.4.2 (elixir_make) 332c649d
  ok
* nerves_runtime 0.6.5 (Hex package) (mix)
  locked at 0.6.5 (nerves_runtime) 7b43e555
  ok
* xyz 0.1.0 (../xyz) (mix)
  ok
* artificery 0.2.6 (Hex package) (mix)
  locked at 0.2.6 (artificery) f6029097
  ok
* distillery 2.0.4 (Hex package) (mix)
  locked at 2.0.4 (distillery) 7fdc6647
  ok
* nerves 1.3.0 (Hex package) (mix)
  locked at 1.3.0 (nerves) 473b95af
  ok
* nerves_toolchain_ctng 1.5.0 (Hex package) (mix)
  locked at 1.5.0 (nerves_toolchain_ctng) 34b8f566
  ok
* nerves_toolchain_arm_unknown_linux_gnueabihf 1.1.0 (Hex package) (mix)
  locked at 1.1.0 (nerves_toolchain_arm_unknown_linux_gnueabihf) ca466a65
  ok
* nerves_system_bbb 1.3.0 (Hex package) (mix)
  locked at 1.3.0 (nerves_system_bbb) 2cc83426
  ok
* shoehorn 0.4.0 (Hex package) (mix)
  locked at 0.4.0 (shoehorn) f3830e22
  ok
```

I can reproduce this reliably with each new test project.

prook, 2018-08-22:
This is also interesting. Instead of arbitrary xyz/test_dep, let's use elixir_ale. Depending on the way the dependency is declared, `mix release` will or won't succeed. If added via hex, everything works as expected, if added using :path, :git or :github, it fails.

This `mix.exs`

```elixir
  #...
  defp deps(target) do
    [
      {:nerves_runtime, "~> 0.6"},
      {:elixir_ale, "~> 1.1"},
    ] ++ system(target)
  end
  #...
```
results in:
```
$ MIX_TARGET=bbb mix do deps.get, release
...
Nerves environment
  MIX_TARGET:   bbb
  MIX_ENV:      dev

==> Assembling release..
==> Building release fw:0.1.0 using environment dev
==> Including ERTS 10.0.4 from /home/prook/.nerves/artifacts/nerves_system_bbb-portable-1.3.0/staging/usr/lib/erlang/erts-10.0.4
==> Generated Shoehorn Boot Script
    Run using shoehorn:
      Interactive: _build/bbb/dev/rel/fw/bin/fw console_boot _build/bbb/dev/rel/fw/bin/shoehorn

==> Packaging release..
Release succesfully built!
```
But this `mix.exs`
```elixir
  # ...
  defp deps(target) do
    [
      {:nerves_runtime, "~> 0.6"},
      {:elixir_ale, github: "fhunleth/elixir_ale"},
    ] ++ system(target)
  end
  # ...
```
fails:
```
$ MIX_TARGET=bbb mix do deps.get, release
...
Nerves environment
  MIX_TARGET:   bbb
  MIX_ENV:      dev

==> Assembling release..
==> Building release fw:0.1.0 using environment dev
==> Including ERTS 10.0.4 from /home/prook/.nerves/artifacts/nerves_system_bbb-portable-1.3.0/staging/usr/lib/erlang/erts-10.0.4
==> Release failed, during .boot generation:
        Undefined applications: [elixir_make]
```

prook, 2018-08-22:
Also interesting! Downgrading from Elixir 1.7.2 to 1.6.6 results in successful build, albeit with a warning:

```
$ MIX_TARGET=bbb mix do clean, deps.get, release
...
==> Assembling release..
==> Building release fw:0.1.0 using environment dev
One or more direct or transitive dependencies are missing from
    :applications or :included_applications, they will not be included
    in the release:

        :elixir_make

    This can cause your application to fail at runtime. If you are sure
    that this is not an issue, you may ignore this warning.

==> Including ERTS 10.0.4 from /home/prook/.nerves/artifacts/nerves_system_bbb-portable-1.3.0/staging/usr/lib/erlang/erts-10.0.4
==> Generated Shoehorn Boot Script
    Run using shoehorn:
      Interactive: _build/bbb/dev/rel/fw/bin/fw console_boot _build/bbb/dev/rel/fw/bin/shoehorn

==> Packaging release..
Release succesfully built!
```

prook, 2018-08-22:
Prying in Distillery [BootScript](https://github.com/bitwalker/distillery/blob/3aa7f686f95fcb39fb826e7a410bd7a162907575/lib/mix/lib/releases/models/boot_script.ex#L145), I can see that `options` differ.

With hex-flavored dependency:

```elixir
> IO.inspect(options)
{:path,
  [
    #...
    '_build/bbb/dev/rel/fw/lib/elixir_make-0.4.2/ebin',
     '/tmp/tmp.UYisY4BkJa/fw_hex/_build/bbb/dev/lib/elixir_make/ebin',
  #...
}
```

These paths are missing in path/git/github style dependency. Subsequently, the call to `:systools.make_script(name, options)` results in `{:error, :systools_make, {:undefined_applications, [:elixir_make]}}`.

This looks like it might be a Distillery issue, but I cannot reproduce it outside Nerves (yet?).

prook, 2018-08-23:
Continuing the investigation, I found out that **order of dependencies matters**. Whenever `elixir_ale` (path/git/github-style dependency declaration) comes before `nerves_runtime` in the dependencies list, `mix firmware`, or `mix release` respectively, succeeds.

prook, 2018-08-23:
Replicated outside Nerves. This is a Distillery issue.

closed 2018-08-23 by prook

prook, 2018-08-23:
Opened bitwalker/distillery#500.