nerves-project/nerves #333: `mix firmware.burn` fails on Windows for unknown reason

tmecklem, 2018-09-05:
### Environment

* Elixir version (`elixir -v`):
> Erlang/OTP 21 [erts-10.0.5] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [hipe]
> Elixir 1.7.2 (compiled with Erlang/OTP 21)

* Nerves environment: (`mix nerves.env --info`)

> $ mix nerves.env --info
> |nerves_bootstrap| Environment Package List
> 
>  Pkg:         nerves_system_br
>  Vsn:         1.4.5
>  Type:        system_platform
>  BuildRunner: nil
> 
>  Pkg:         nerves_system_rpi0_zbar
>  Vsn:         1.4.0
>  Type:        system
>  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}
> 
>  Pkg:         nerves_toolchain_armv6_rpi_linux_gnueabi
>  Vsn:         1.1.0
>  Type:        toolchain
>  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}
> 
>  Pkg:         nerves_toolchain_ctng
>  Vsn:         1.5.0
>  Type:        toolchain_platform
>  BuildRunner: nil
> 
> |nerves_bootstrap| Loadpaths Start
> 
> 
> Nerves environment
>  MIX_TARGET:   rpi0_zbar
>  MIX_ENV:      dev
> 
> |nerves_bootstrap| Environment Variable List
>  target:     rpi0_zbar
>  toolchain:  /home/tv/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-linux_x86_64-1.1.0
>  system:     /home/tv/.nerves/artifacts/nerves_system_rpi0_zbar-portable-1.4.0
>  app:        /mnt/c/Code/Github/GregMefford/omni_eye/eye_fw
> 
> |nerves_bootstrap| Loadpaths End

* Running inside WSL with fwup installed in WSL and on the Windows side (binary only, not using chocolately). This was an issue experienced by one of the conference attendees using a an IBM Thinkpad laptop running Windows 10.

### Current behavior

Running `mix firmware.burn` enumerates the SD card device list correctly, but once one is chosen an error appears when trying to write to the card:

```bash
$ mix firmware.burn

Nerves environment
 MIX_TARGET:   rpi0_zbar
 MIX_ENV:      dev

Use 7.29 GiB memory card found at \\.\PhysicalDrive1? [Yn]
Start-Process : This command cannot be run due to the error: The system cannot find the file specified.
At line:1 char:1
+ Start-Process fwup -ArgumentList '-a -i C:\Code\Github\GregMefford\om ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   + CategoryInfo          : InvalidOperation: (:) [Start-Process], InvalidOperationException
   + FullyQualifiedErrorId : InvalidOperationException,Microsoft.PowerShell.Commands.StartProcessCommand
```

Running the following does work in a PowerShell window:
`Start-Process fwup -ArgumentList '-a -i C:\Code\Github\GregMefford\omni_eye\eye_fw\_build\rpi0_zbar\dev\nerves\images\eye_fw.fw -t complete -d \\.\PhysicalDrive1'`

### Expected behavior

The SD card should be written with the firmware.

tmecklem, 2018-09-05:
@michaelkschmidt have you run into this issue before? It's difficult to tell what's happening and we may have some time to troubleshoot before the end of the day once the free lab time starts.

michaelkschmidt, 2018-09-05:
Is fwup.exe in the path on Windows?  Its acting like it can't find it.  There might be a System Path vs User Path issue

michaelkschmidt, 2018-09-05:
It looks like we needed to reboot to update the System Path.

Note it takes a while to start--we copy the firmware to a known location so fwup.exe can find it.  Not sure if this is the delay but we need to figure out why its taking so long

fhunleth, 2020-11-14:
Closing since this doesn't seem to be an issue with WSL 2.

closed 2020-11-14 by fhunleth