nerves-project/nerves #341: Setting MIX_TARGET=rpi3 and forgetting to depend on nerves_system_rpi3 fails unintuitively

fhunleth, 2018-09-16:
The failure is that when a C program is being compiled, the host's compiler is called to build it. There are hints that something is wrong since NERVES_SYSTEM and NERVES_TOOLCHAIN get printed as being unset. However, the compiler errors from C programs can be long enough to hide those prints. 

I'm wondering if there's a way to catch this earlier. For example, is it possible to warn if MIX_TARGET isn't set to host and there's not a Nerves system in the deps list? I know that we try not to have an MIX_TARGET references outside of the mix.exs so maybe there's a better way.

This took me a little bit to catch so I'm hoping there's some way to make it easier for next time.

Hermanverschooten, 2018-10-28:
Frank, how do you get in this situation?

If I comment out the `++ system(target)` in`mix.exs`, I get the following error:
```
Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

NERVES_SYSTEM is unset
NERVES_TOOLCHAIN is unset
** (Mix) Environment variable $NERVES_SYSTEM is not set.

This variable is usually set for you by Nerves when you specify the
$MIX_TARGET. For examples please see
https://hexdocs.pm/nerves/getting-started.html#create-the-firmware-bundle
```

fhunleth, 2018-10-28:
That seems like that should do it. I entered this when I was debugging someone else's project and never got a chance to make a minimal example. They had some C code that would get built in a dependency, but the wrong compiler was being called. I wonder if this can be closed. @mobileoverlord?

closed 2020-10-03 by fhunleth

layeddie, 2020-11-22:
Hope I am referencing this issue correctly. I did a search for "$NERVES_SYSTEM is not set" and found this issue. I can ssh into my pi from my mac.

MIX_TARGET loses setting to rpi3 and becomes MIX_TARGET:   host

my terminal output follows...

`Eddies-MacBook-Pro:nerves_heartbeat elay15$ MIX_TARGET=rpi3 mix firmware.gen.script
==> nerves
==> nerves_heartbeat

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

OK to overwrite upload.sh? [Yn] y
Writing upload.sh...

Eddies-MacBook-Pro:nerves_heartbeat elay15$ mix firmware

Nerves environment
  MIX_TARGET:   host
  MIX_ENV:      dev

NERVES_SYSTEM is unset
NERVES_TOOLCHAIN is unset
** (Mix) Environment variable $NERVES_SYSTEM is not set.

This variable is usually set for you by Nerves when you specify the
$MIX_TARGET. It is unusual to need to specify it yourself.

Some things to check:

1. In your `mix.exs`, is the value that you have in $MIX_TARGET in the
  `@all_targets` list? If you're not using `@all_targets`, then the
  $MIX_TARGET should appear in the `:targets` option for `:nerves_runtime`
  and other packages that run on the target.

2. Do you have a dependency on a Nerves system for the target? For example,
  `{:nerves_system_rpi0, "~> 1.8", runtime: false, targets: :rpi0}`

3. Is there a typo? For example, is $MIX_TARGET set to `rpi1` when it should
  be `rpi`.

4. Is there a typo in the package name of the system? For example, if you
  have a custom system, `:nerves_system_my_board`, does the spelling of the
  system in the dependency in your `mix.exs` match the spelling in your
  system project's `mix.exs`?

For build examples in the Nerves documentation, please see
https://hexdocs.pm/nerves/getting-started.html#create-the-firmware-bundle

Just got this error when trying to deploy update.sh to rpi3b`

Hermanverschooten, 2020-11-22:
Generating the `upload.sh` script with the `MIX_TARGET` set, does not mean that is bound to that target. You need to specify `MIX_TARGET` every time, or do an `export MIX_TARGET=rpi3`, while you remain in that terminal session.

layeddie, 2020-11-22:
I found that it did not like this command MIX_TARGET=rpi3 mix firmware.gen.script
changed to 
export MIX_TARGET=rpi3
mix firmware.gen.script

Then it was OK.

Hermanverschooten, 2020-11-22:
You know you only need to do that once?  It generates the `upload.sh` script. To generate the firmware, just do `mix firmware`, either prefixed with `MIX_TARGET=rpi3` or with it exported. You can then use `upload.sh` to send it to a connected nerves device, first time you burn in it to the SD-card directly using `mix firmware.burn`, again either prefixed or with the target exported. The easiest is to export the target, then you do no need to prefix it all the time.

layeddie, 2020-11-22:
So you mean only need to do the export MIX_TARGET=rpi3 once
then
mix firmware.gen.script
then
./upload.sh
thats the order that seems to work.

when I used MIX_TARGET=rpi3 mix firmware.gen.script it did not seem to export the MIX_TARGET=rpi3 correctly. So I changed to
export MIX_TARGET=rpi3
then
mix firmware.gen.script (asks me to overwrite upload.sh)
then
./upload.sh
this re-initializes the pi with the new code

I was following this blog
http://www.ipushbuttons.io/blog/getting-started-elixir-nerves-part-2/

Hermanverschooten, 2020-11-23:
You only need to generate the script once, next you use `mix firmware` to build your firmware and `./upload.sh` to upload it to your device, or `mix firmware.burn` if you are writing it to the SD card directly. And you need to do the export only once per terminal session.

1. export MIX_TARGET
2. mix firmware
3. mix firmware.burn
4. put the SD card in, boot and test
5. change code
6. mix firmware burn
7. .upload.sh
and repeat 5-7

layeddie, 2020-11-23:
Hello Herman,
Thanks for clarifying that.
Eddie Lay