nerves-project/nerves #343: Should mix firmware.burn call mix firmware

fhunleth, 2018-09-25:
Currently `mix firmware.burn` doesn't rebuild. I recently watched this confuse two people into thinking they had installed the latest and greatest on a MicroSD card when, in fact, they hadn't rebuilt. I'm accustomed to this behavior, but I suspect that it's not expected. What do others think?

ConnorRigby, 2018-09-25:
I've always been wondered about this. I personally prefer this because i'm accustomed to it also, but i can see the benefits of changing it. 

Another option could be to add a `--compile` (?) flag to the `firmware.burn` task

mobileoverlord, 2018-09-25:
If the compilation is stale at this time, then mix.firmware would have to also be invoked, we could probably check if sources are stale and halt with an error that they would need to call `mix firmware`

mobileoverlord, 2018-09-25:
An issue is that we do not link project source -> fw file, so we don't really know if the contents of the fw is stale form the sources. So we would have to call `mix firmware` on every call to `mix firmware.burn`

fhunleth, 2018-09-25:
My preference would be that `mix firmware.burn`'s default would be to check whether compilation is stale since it's a power user feature to skip the step to save time. Also, fwiw, I call `fwup` directly to save time since just starting `mix` to do anything has always felt slow to me.

joelbyler, 2018-10-11:
It might be really confusing for folks who are accustomed to using `mix firmware.burn` at this point but I'm wondering if the name is what's confusing folks.  Perhaps since `firmware` will build firmware, they expect `firmware.burn` to do that plus write to an sd card?  What if we change `firmware.burn` to do both and rename the current task to `burn`?  

**ps** I've been super busy but #hacktoberfest is a good reminder for me to come back and check in on y'all 😄 Let me know if there's any way I can help out, I'm glad to help if there's a need.

PR #356 (2018-10-18): Call mix firmware from firmware.burn task.
Also moving original firmware.burn task into a new burn task, which only
burns firmware to an SDCard in case folks want to continue working that
way. This is being done in an effort to avoid confusion for new nerves
users as we've noticed that some users misinterpret this task and think
that it does both.  Hopefully this change will make the tasks more
intuitive based on their task names.

Was initially intended to resolve #343 but submitting as an alternative approach which retains the functionality of the old task in case its still desired by some u [... cut]
closed 2018-10-15 by mobileoverlord