nerves-project/nerves #350: In getting started, add simple tutorial for connecting the first time

fhunleth, 2018-10-06:
The getting started guide ends with building firmware, but what do to do next is less clear. While we support a few platforms, I think that walking through the Raspberry Pi Zero (usb gadget) or Raspberry Pi 3 (Ethernet/WiFi) could cover most users.

fhunleth, 2018-10-12:
Right, that wasn't quite was I was going for with this issue, but what you're proposing sounds really valuable. We could mold it to work for other examples. The hardest part is always getting the first one, though, and "hello_network" certainly is a popular example.

GregMefford, 2018-10-12:
I'm happy to collaborate on that as well, @joelbyler! Just get in touch if you get stuck or want to bounce ideas around. ❤️ 🚀

deanchouinard, 2019-03-30:
I put together a [draft](https://github.com/deanchouinard/nerves/commit/869ec270d6410647a99d850107a7297ed21be6c8) based on some of the questions I had when first starting and seeing some of the questions you guys answer on Slack. Let me know what you think. I'm certainly open to any changes.

fhunleth, 2022-01-14:
The PR associated with this was merged long ago. Somehow we missed closing it.

closed 2022-01-14 by fhunleth