nerves-project/nerves #351: Unable to mount root filesystem

MatteoJoliveau, 2018-10-08:
### Environment

* Elixir version (`elixir -v`):
```
Erlang/OTP 21 [erts-10.1] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:1] [hipe]
Elixir 1.7.3 (compiled with Erlang/OTP 19)
```
* Nerves environment: (`mix nerves.env --info`):
```
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.5.2
  Type:        system_platform
  BuildRunner: nil

  Pkg:         nerves_system_rpi3
  Vsn:         1.5.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: nil

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi3
  toolchain:  /Users/matteo/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-darwin_x86_64-1.1.0
  system:     /Users/matteo/.nerves/artifacts/nerves_system_rpi3-portable-1.5.0
  app:        /Users/matteo/Software/miskappa/iot

|nerves_bootstrap| Loadpaths End
```

* Working from a 2017 MacBook Pro 13", macOS 10.14 Mohave, targeting a Raspberry Pi 3 B v1.2.
Using stock `fwup.conf` and `erlinit.config`.

### Current behavior

It fails to boot with errors:
```
EXT4-fs VFPS: Can't find ext4 filesystem
erlinit: Cannot mount /dev/mmcblk0p3 at /root: invalid argument
```

### Expected behavior
It should mount the root filesystem

I couldn't manage to find any issue like that online, so I'm resorting to asking directly. If I should open this issue on another project please let me know.

mobileoverlord, 2018-10-08:
This is expected on the first boot of the system. The application data partition is formatted the first time around. Are you only encountering this on first boot, or are you getting this on subsequent boots?

MatteoJoliveau, 2018-10-08:
I'm constantly getting this error, and the boot simply fails with various other messages like `blk_partition_remap fail for partition 3`.

I'm trying to grab a full dump to attach

MatteoJoliveau, 2018-10-08:
Which is easier said than done since it keeps shutting itself down after a bunch of `blk_partition_remap fail`.

Btw both the board and the SD Card are working correctly with Raspbian

mobileoverlord, 2018-10-08:
I just created and successfully booted the rpi3 from a new project. It seems that you might be upgrading from a previous version. If you have access to a serial console cable, it would probably be easier to grab the output to send over. Also, is this project posted publicly anywhere that we could take a look?

MatteoJoliveau, 2018-10-08:
The project is a simple blinker that I bootstrapped this morning using `nerves.new`. No upgrades at all.

I'll try to boot the Blinker example directly and see if it works, otherwise I'll upload my code.

MatteoJoliveau, 2018-10-08:
Just tried with [nerves_examples/blinky](https://github.com/nerves-project/nerves_examples), I managed to get a long sequence of `SQUASHFS error: Unable to read page, block b396e6, size 1772e`. Fresh clone from GitHub, `mix deps.get`, `mix firmware` and `mix firmware.burn` as per documentation.

ConnorRigby, 2018-10-08:
This is probably a dead or dying SDcard or the card was removed/slipped out of the cradle.

MatteoJoliveau, 2018-10-08:
@ConnorRigby I thought so, but it works fine with Raspbian and passes all health checks. Also I've been trying different MicroSD adapters and none seemed to make any difference

MatteoJoliveau, 2018-10-09:
New SD card, seems to have fixed it so I confirm the supposedly bad card.

Still weird since it works with any other system I flashed on it, but I won't argue since now it's working.
Thanks, everyone!

closed 2018-10-09 by MatteoJoliveau

lucaong, 2018-10-20:
I have a very similar, if not identical issue. I have an application running on Nerves on Raspberry Pis Zero W, which I am maintaining and updating regularly since a year. After my last update, I am not able to boot anymore. The ACT LED just blinks in patterns of 2, which seems to indicate a problem booting from the SD card.

I obviously initially blamed my own code, so I burned the Nerves example app `blinky` on the SD card: still cannot boot. Then I assumed it was the SD card, but other SD cards, even brand new, behave exactly the same. Finally, I tried with a different Pi, and still got the same result.

One thing I might think of, but just shooting in the dark, I recently upgraded my Mac to Mojave. @MatteoJoliveau what is your development setup?

lucaong, 2018-10-20:
oh sorry, I misread, I see now in @MarceloAlves case it was a faulty SD card. Sorry for the noise

lucaong, 2018-10-20:
Just for completeness, also in my case the problem was that both SD cards were faulty. A third one, from a different brand, works (insert facepalm). Sorry again for the noise!