nerves-project/nerves #363: mix nerves.system.shell throws an error on Logger.Backends.Console

Hermanverschooten, 2018-11-08:
### Environment

* Elixir version (`elixir -v`): 1.7.3
* Nerves environment: x86_64 v1.5.1

### Current behavior

After cloning the repo for branch v1.5.1, running
 `mix nerves.system.shell`

gives:

```
:gen_event handler Logger.Backends.Console installed in Logger terminating
** (exit) an exception was raised:
** (RuntimeError) no device registered with the name :user
                                                                                                                                                                                  (logger) lib/logger/backends/console.ex:160: Logger.Backends.Console.async_io/2
(logger) lib/logger/backends/console.ex:145: Logger.Backends.Console.log_event/5
(logger) lib/logger/backends/console.ex:49: Logger.Backends.Console.handle_event/2
(stdlib) gen_event.erl:577: :gen_event.server_update/4
(stdlib) gen_event.erl:559: :gen_event.server_notify/4
(stdlib) gen_event.erl:300: :gen_event.handle_msg/6
(stdlib) proc_lib.erl:249: :proc_lib.init_p_do_apply/3
```

@ConnorRigby tells me it can be ignored due to a weird elixir version mismatch, to @fhunleth it seems worth an issue.

fhunleth, 2018-11-08:
@ConnorRigby What do you mean by weird elixir version mismatch?

ConnorRigby, 2018-11-08:
It shouldn't actually break anything being used. 

I've hit this issue outside of Docker with certain combinations of OTP/Elixir that i've never really be able to track down. I believe i remember seeing it with elixir 1.4 and otp 20 or 21. I'm not sure the _actual_ cause of the error, only that i've always fixed it by shuffling versions around

GregMefford, 2018-11-11:
I don't believe this is a Docker / Nerves System Shell issue.
You can replicate this error message by installing Elixir 1.7.3 (built against OTP 20) and OTP 21.

I believe it's because in OTP 21, Erlang switched to using the new Erlang `:logger` module by default, to it breaks Elixir's `Logger` integration with `:gen_event`-handling.

The is to use an Elixir version that was built against OTP 21 for it to work (`1.7.3-otp-21` if using `asdf`) if you're running on OTP 21.

fhunleth, 2018-11-13:
Makes sense. I'll close. 

@Hermanverschooten - thanks for posting. I'm sure that others will run into this and appreciate Greg's explanation here.

closed 2018-11-13 by fhunleth

Hermanverschooten, 2018-11-21:
@fhunleth Sorry for the late reply, but since this is happening inside the dockerized environment does this not suggest that there is a version issue there?

GregMefford, 2018-11-21:
That's a good question, but I think in this case, the error is thrown by the environment on your host before it invokes the Docker command. I believe this is the case because I can make the error go away by using an Elixir (on my host) that's compiled against OTP 21. Please let me know if you're seeing something different or that isn't resolving it for you.