nerves-project/nerves #375: to stderr

henrik-jensen, 2019-01-10:
### Environment

* Elixir version (`elixir -v`):
* Nerves environment: (`mix nerves.env --info`)
* Additional information about your host, target hardware or environment that
  may help

### Current behavior

Include errors, stacktraces, screenshots or code that may help us reproduce the
issue.

### Expected behavior

GregMefford, 2019-01-10:
Was this submitted without finishing the details? It's not clear from the title what it's about.

henrik-jensen, 2019-01-10:
Oh I'm sorry. That was an error.
I was reading issue #347 and then closing the tab .
I think I "pocket dialed" it (ctrl+T and some unfortunate github shortcuts) but I didn't notice I actually created an issue. Sorry again, closing it now.

closed 2019-01-10 by henrik-jensen